cocotb/cocotb #3172: Mismatch between `cocotb.triggers.Timer` and Verilog default delay units

shuckc, 2023-01-05:
It would be helpful if `cocotb.triggers.Timer` could precisely match unit-less verilog delays, which are multiples of the simulator time unit. For instance to write assertions on the value of `q` that are independent of the simulator timescale:  

```verilog
module top ();
  reg [4:0] q = 0;
  initial begin
    #100; q <= 1;
    #100; q <= 2;
    #100; q <= 3;
  end
endmodule
```

Calling `await Timer(100)`, inheriting default kwarg `units="step"` interprets the `time` argument as a multiple of the simulator *precision* not *time unit*. This does not match Verilog's default choice. 

By default [cocotb Makefiles set timescale](https://github.com/cocotb/cocotb/blob/5f40366c448a1828d40fe6d07919788bbb71fbdd/cocotb/share/makefiles/Makefile.inc#L91) `1ns/1ps`, so knowing this we can write a testbench that passes with `make SIM=icarus`.

```python
import cocotb
from cocotb.triggers import Timer, NextTimeStep

@cocotb.test()
async def delay_units(dut):
    await Timer(95*1000)
    assert dut.q.value == 0
    await Timer(10*1000)
    assert dut.q.value == 1
    await Timer(100, units="ns")
    assert dut.q.value == 2
    await Timer(100*1000, units="step")
    assert dut.q.value == 3
```

However changing simulator timeunit or precision triggers different failures:

*  `make clean && make COCOTB_HDL_TIMEUNIT=1ps` results in `1ps/1ps`, so fails at the first assertion 3==0
    as `#100` is 100ps and python has waited for 95ns, missing the whole waveform.

*  `make clean && make COCOTB_HDL_TIMEUNIT=1us` results in `1us/1ps`, so fails at the 2nd assertion 0==1
    as `#100` is 1us and python has waited 95ns, and is yet to see the change value.

* `make clean && make COCOTB_HDL_TIMEPRECISION=1ns` results in `1ns/1ns`, so fail on the first assertion 3==0
    as `#100` is 100ns, however python has waited for *95us* and missed whole waveform.

In short, it would be great to add `await Timer(100, units="timeunit")` that would allow the python testbench to pass independent of the simulator timescale, although I accept it is probably too late to change the default:

```python
@cocotb.test()
async def delay_units(dut):
    await Timer(95, units="timeunit")
    assert dut.q.value == 0
    await Timer(10, units="timeunit")
    assert dut.q.value == 1
    await Timer(100, units="timeunit")
    assert dut.q.value == 2
    await Timer(100, units="timeunit")
    assert dut.q.value == 3
 ```
 
I will see if I can raise a PR

imphil, 2023-01-11:
Thanks for opening this issue, @shuckc. 

Things to note here:
* The name "step" for "timeprecision" is matching the SV terminology: "The step time unit is equal to the global time precision. Unlike other time units, which represent physical units, a step cannot be used to set or modify either the precision or the time unit." (Section 3.14.3 Simulation time unit of the SV 2012 LRM).
* It is a mismatch between a SV delay statement and cocotb Timer() that the default of of the "units" argument in Timer is "steps" and not the timeunit.
* cocotb is missing functionality to advance time by one timeunit.

So I see a number of things we can do:
* Introduce a new way to advance time by timeunits, as you mentioned. This would be a strict addition to the current set of functionality, so there's no backwards-compat issue.
* It is too late to change the default value of Timer() from "steps" (timeprecision) to "timeunits". But: we have deprecated relying on the default value anyways, and by having users explicitly specify the unit they want we can avoid this point of confusion as well.

Going forward:
* Would you mind opening a PR with an implementation of the "timeunits" (or whatever we might want to call it) unit?
* What is VHDL doing, and what terminology are they using? I'm less familiar with that, so let's make sure we double-check.

shuckc, 2023-01-17:
Thanks for your considered reply @imphil 

I've wired up support for `Timer(..., units="timeunit")` through the python interface. The implementation needs information from the simulator interface that is not currently exposed. Adding it to `simulatormodule.cpp` is straightforward, as are the `vpi` and `vhpi` backends. After reading the Modelsim `fli` documentation I cannot find any functions that expose the equivalent behaviour, so that remains NYI. Am I correct that `gpi` is an internal interface therefore it does not need to maintain abi compatibility with any other code outside of the repository?

When I tried the obvious implementation against Icarus, I found the vpi call  `*timeunit = vpi_get(vpiTimeUnit, NULL);` returns the precision (???), ie the same as calling with argument `vpiTimePrecision`. I think the cause is that a scope needs to be provided as the root scope has a timeunit equal to the precision (defaulted here https://github.com/steveicarus/iverilog/blob/master/vvp/vpi_scope.cc#L592). I will look into it further.

It occurs to me that there is no guarantee that this information is available in all simulators - it would be fairly reasonable for a simulator to translate the verilog delay units to multiples of the precision much earlier (during parsing) to, so by the time the vpi interface is hooked up the necessary information might not be in scope.

marlonjames, 2023-01-17:
VHDL has the concept of resolution limit. The default is `1 fs`, but simulators may allow selecting a secondary unit of type `TIME`, which would be `1 ps`, `1 ns`, etc.
Any `TIME` value less than the resolution limit is set to `0`.

So for VHPI, only the resolution limit / `step` unit is applicable. `timeunit` could act the same as `step` for the VHDL case.

imphil, 2023-01-18:
Thanks @shuckc for having a look. Your reply raises an interesting point: In Verilog, the meaning of timeunit is local and can be influenced by things like the timescale directive (that's why the scope has to be provided in `vpi_get()`). 

Now things become interesting: what would be the right scope when writing `Timer(1, units="timeunit")` in a Python testbench? The Python code has its own scoping rules, so using an arbitrary Verilog scope seems wrong. We could use the global timeunit as default, for sure, but that's no guarantee that we match the behavior a user would get when they'd put a `#1` delay somewhere.

What was the behavior you were thinking about?

shuckc, 2023-01-19:
It seems my understanding was incorrect. I thought Icarus enforced all verilog files use the *same* timescale with warnings, to choose a global precision/timeunit (that we could fetch). Instead it checks all files include a timescale, or all inherit the default (from a command file), and that a timescale does not flow from one file to the next. It then chooses a global precision, but the timeunits indeed remain scoped and can differ by module.

I've verified this by editing toy modules with different timescales, elaborating them with `iverilog` and then inspecting the input to vvp:

```
$ head -n5 *.v
==> a.v <==
`timescale 1ns/1ps
module top ();
  reg [4:0] q = 0;
  initial begin
    #100; q <= 1;

==> b.v <==
`timescale 1us/1ns
module topb ();
  reg [4:0] q = 0;
  initial begin
    #100; q <= 1;
```
Then:
```
$ iverilog a.v b.v -s top -s topb -Wall -g2012
$ cat a.out
#! /usr/local/bin/vvp
:ivl_version "12.0 (devel)" "(s20150603-1242-gbc0fc4ab)";
:ivl_delay_selection "TYPICAL";
:vpi_time_precision - 12;
:vpi_module "/usr/local/lib/ivl/system.vpi";
:vpi_module "/usr/local/lib/ivl/vhdl_sys.vpi";
:vpi_module "/usr/local/lib/ivl/vhdl_textio.vpi";
:vpi_module "/usr/local/lib/ivl/v2005_math.vpi";
:vpi_module "/usr/local/lib/ivl/va_math.vpi";
:vpi_module "/usr/local/lib/ivl/v2009.vpi";
S_0x565439ebb6b0 .scope package, "$unit" "$unit" 2 1;
 .timescale 0 0;
S_0x565439ebb840 .scope module, "top" "top" 3 2;
 .timescale -9 -12;
v0x565439e92e20_0 .var "q", 4 0;
S_0x565439e81ce0 .scope module, "topb" "topb" 4 2;
 .timescale -6 -9;
v0x565439e93280_0 .var "q", 4 0;
```
The global precision is -12 (ps). You can see three module scopes, `top` is `1ns/1np`, `topb` is `1us/1ns`, the `$unit` scope seems to be `1s/1s` (the default). 

As an interesting vvp implementation note, `#delays` get elaborated to multiples of the precision, rather than multiples of the scope's timeunit. Both modules use `#100` per my original post, and have been translated:
```
...
    .scope S_0x565439ebb840;
T_1 ;
    %delay 100000, 0;
    %pushi/vec4 1, 0, 5;
...
.scope S_0x565439e81ce0;
T_3 ;
    %delay 100000000, 0;
    %pushi/vec4 1, 0, 5;
```
It seems that the timescale values on the scope are there only for vpi to inspect, they probably play no part in the actual simulation.

So in terms of this issue, given we can have multiple top units with different time units, and making a choice on behalf of the user is at best arbitrary. I will have a look if it's possible to do the following, and have them convert to the appropriate steps:

```python
await Timer(100, units="timeunit", scope="top")
...
await Timer(100, units="timeunit", scope="topb")
```
If not I will close the issue.

imphil, 2023-01-19:
The scope of a timeunit can be a package, an interface, a module, a compilation unit, and a couple more things. So specifying only the toplevel in your "scope=" example above won't do the trick. Timeunits can be set either with the `timescale` directive, or using the `timeunit` and `timeprecision` keywords.

Relevant SV-2012 LRM sections: 
* 3.14.2.2 The timeunit and timeprecision keywords
* 3.14.2.3 Precedence of timeunit, timeprecision, and `timescale

So instead of taking the timeunit from the simulation and answering the question "from which scope should we take it" (which, I think, doesn't have a good answer), we could invent our own scoping and "timeunit" definitions within cocotb.

So if users really want, they could write (hypothetical)

```python3
cocotb.define_timeunit(100, units="steps")

await Timer(100, units="timeunits")
```

Now, I think that's probably confusing users more than helping, so we'd need to have a closer look at the use cases actually.