cocotb/cocotb #3181: Is there a way to disable writing COCOTB_RESULTS_FILE output?

forensicgarlic, 2023-01-11:
I'm using the latest cocotb on 64 bit linux, but running in a test environment where the base filesystem is read-only. Thus cocotb tests run fine until they try to write their results.xml file output, where they find the file system is READ-ONLY and crash. Is there a way to disable writing the results.xml file? Or alternatively, can a provide a path in COCOTB_RESULTS_FILE to a part of the file system that's not read-only?

here's the exact error
 ```
ERROR: called callback function threw exception
-Info: /build/.../cocotb/examples/rtl/src/adder.sv:29: $dumpvar ignored, as Verilated without --trace
Traceback (most recent call last):
  File "/build/.../python3.9/xml/etree/ElementTree.py", line 762, in _get_writer
    write = file_or_filename.write
AttributeError: 'str' object has no attribute 'write'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/build/.../cocotb/1/7/1/dist/lib/python3.9/cocotb/scheduler.py", line 348, in _test_completed
    self._handle_result(test)
  File "/build/.../python/cocotb/1/7/1/dist/lib/python3.9/cocotb/regression.py", line 339, in _handle_result
    self._execute()
  File "/build/.../python/cocotb/1/7/1/dist/lib/python3.9/cocotb/regression.py", line 532, in _execute
    return self._tear_down()
  File "/build/.../python/cocotb/1/7/1/dist/lib/python3.9/cocotb/regression.py", line 290, in _tear_down
    self.xunit.write()
  File "/build/.../python/cocotb/1/7/1/dist/lib/python3.9/cocotb/xunit_reporter.py", line 80, in write
    ET.ElementTree(self.results).write(self.filename, encoding="UTF-8")
  File "/build/.../dist/lib/python3.9/xml/etree/ElementTree.py", line 732, in write
    with _get_writer(file_or_filename, enc_lower) as write:
  File "/build/.../dist/lib/python3.9/contextlib.py", line 119, in __enter__
    return next(self.gen)
  File "/build/.../dist/lib/python3.9/xml/etree/ElementTree.py", line 768, in _get_writer
    file = open(file_or_filename, "w", encoding=encoding,
OSError: [Errno 30] Read-only file system: 'results.xml'
- :0: Verilog $finish
- /build/...cocotb/1/7/1/dist/lib/python3.9/cocotb/share/lib/verilator/verilator.cpp:118: Verilog $finish
- /build/.../cocotb/1/7/1/dist/lib/python3.9/cocotb/share/lib/verilator/verilator.cpp:118: Second verilog $finish, exiting
/build/build/.../cocotb/sh/test/sh_vlt_test_adder.sh
```

imphil, 2023-01-11:
You should be able to pass an arbitrary absolute path as COCOTB_RESULTS_FILE, e.g. `COCOTB_RESULTS_FILE=/your/writeable/mount/results.xml`, or, if nothing is writable, set `COCOTB_RESULTS_FILE=/dev/null` (which I'm not sure is valuable, but should work).

forensicgarlic, 2023-01-12:
that worked, thanks!

closed 2023-01-12 by forensicgarlic