cocotb/cocotb #3183: CI: msvc/icarus tests are failing on master

imphil, 2023-01-12:
https://github.com/cocotb/cocotb/actions/runs/3901327584/jobs/6663044237

Our lovely conda problem keeps coming back, last "fixed" with https://github.com/cocotb/cocotb/issues/3166.

```
nox > pytest -v --cov=cocotb --cov-branch --cov-report= -k 'not simulator_required'
Traceback (most recent call last):
  File "C:\Miniconda\lib\runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Miniconda\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "D:\a\cocotb\cocotb\.nox\dev_test_nosim\Scripts\pytest.EXE\__main__.py", line 7, in <module>
    sys.exit(console_main())
  File "D:\a\cocotb\cocotb\.nox\dev_test_nosim\lib\site-packages\_pytest\config\__init__.py", line 190, in console_main
    code = main()
  File "D:\a\cocotb\cocotb\.nox\dev_test_nosim\lib\site-packages\_pytest\config\__init__.py", line 148, in main
    config = _prepareconfig(args, plugins)
  File "D:\a\cocotb\cocotb\.nox\dev_test_nosim\lib\site-packages\_pytest\config\__init__.py", line 329, in _prepareconfig
    config = pluginmanager.hook.pytest_cmdline_parse(
  File "D:\a\cocotb\cocotb\.nox\dev_test_nosim\lib\site-packages\pluggy\_hooks.py", line 265, in __call__
    return self._hookexec(self.name, self.get_hookimpls(), kwargs, firstresult)
  File "D:\a\cocotb\cocotb\.nox\dev_test_nosim\lib\site-packages\pluggy\_manager.py", line 80, in _hookexec
    return self._inner_hookexec(hook_name, methods, kwargs, firstresult)
  File "D:\a\cocotb\cocotb\.nox\dev_test_nosim\lib\site-packages\pluggy\_callers.py", line 55, in _multicall
    gen.send(outcome)
  File "D:\a\cocotb\cocotb\.nox\dev_test_nosim\lib\site-packages\_pytest\helpconfig.py", line 103, in pytest_cmdline_parse
    config: Config = outcome.get_result()
  File "D:\a\cocotb\cocotb\.nox\dev_test_nosim\lib\site-packages\pluggy\_result.py", line 60, in get_result
    raise ex[1].with_traceback(ex[2])
  File "D:\a\cocotb\cocotb\.nox\dev_test_nosim\lib\site-packages\pluggy\_callers.py", line 39, in _multicall
    res = hook_impl.function(*args)
  File "D:\a\cocotb\cocotb\.nox\dev_test_nosim\lib\site-packages\_pytest\config\__init__.py", line 1058, in pytest_cmdline_parse
    self.parse(args)
  File "D:\a\cocotb\cocotb\.nox\dev_test_nosim\lib\site-packages\_pytest\config\__init__.py", line 1346, in parse
    self._preparse(args, addopts=addopts)
  File "D:\a\cocotb\cocotb\.nox\dev_test_nosim\lib\site-packages\_pytest\config\__init__.py", line 1229, in _preparse
    self.pluginmanager.load_setuptools_entrypoints("pytest11")
  File "D:\a\cocotb\cocotb\.nox\dev_test_nosim\lib\site-packages\pluggy\_manager.py", line 287, in load_setuptools_entrypoints
    plugin = ep.load()
  File "C:\Miniconda\lib\importlib\metadata.py", line 77, in load
    module = import_module(match.group('module'))
  File "C:\Miniconda\lib\importlib\__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "D:\a\cocotb\cocotb\.nox\dev_test_nosim\lib\site-packages\_pytest\assertion\rewrite.py", line 168, in exec_module
    exec(co, module.__dict__)
  File "D:\a\cocotb\cocotb\.nox\dev_test_nosim\lib\site-packages\pytest_cov\plugin.py", line 6, in <module>
    import coverage
  File "D:\a\cocotb\cocotb\.nox\dev_test_nosim\lib\site-packages\coverage\__init__.py", line 15, in <module>
    from coverage.control import Coverage, process_startup
  File "D:\a\cocotb\cocotb\.nox\dev_test_nosim\lib\site-packages\coverage\control.py", line 28, in <module>
    from coverage.collector import Collector, HAS_CTRACER
  File "D:\a\cocotb\cocotb\.nox\dev_test_nosim\lib\site-packages\coverage\collector.py", line 19, in <module>
    from coverage.data import CoverageData
  File "D:\a\cocotb\cocotb\.nox\dev_test_nosim\lib\site-packages\coverage\data.py", line 24, in <module>
    from coverage.sqldata import CoverageData
  File "D:\a\cocotb\cocotb\.nox\dev_test_nosim\lib\site-packages\coverage\sqldata.py", line 18, in <module>
    import sqlite3
  File "C:\Miniconda\lib\sqlite3\__init__.py", line 23, in <module>
    from sqlite3.dbapi2 import *
  File "C:\Miniconda\lib\sqlite3\dbapi2.py", line 27, in <module>
    from _sqlite3 import *
ImportError: DLL load failed while importing _sqlite3: The specified module could not be found.
nox > Command pytest -v --cov=cocotb --cov-branch --cov-report= -k 'not simulator_required' failed with exit code 1
nox > Session dev_test_nosim failed.
Error: Process completed with exit code 1.
```

FYI @themperek

themperek, 2023-01-18:
Seems to be OK now?

imphil, 2023-01-18:
Looks like it. The next time this happens I'm throwing out those tests, it's just not a reliable base to be testing from. *argh*