cocotb/cocotb #3185: Add VERILOG_INCLUDE_DIRS support to Makefiles

davekeeshan, 2023-01-12:
It would be a nice feature to have the makefiles able to take in a list of directories that are part of the include list and not the source, and format them correctly for each tool.  I am working between icarus and xcelium, they pass in include dirs in a different way.  

This is obviously solvable via just the `COMPILE_ARGS +=` for each tool, but this is a bit cludgy.

I added a few dirs to my common Makefile like this:
```
VERILOG_INCLUDE_DIRS = \
	../verilog/common \
	../verilog/fft 

```
`VERILOG_INCLUDE_DIRS` was a name format already in use in the questa makefile

I added this to the xcelium, ius and verilator makefiles:
```
ifdef VERILOG_INCLUDE_DIRS
    COMPILE_ARGS += $(addprefix +incdir+, $(VERILOG_INCLUDE_DIRS))
endif
```

and I added this to the icarus one:
```
ifdef VERILOG_INCLUDE_DIRS
    COMPILE_ARGS += $(addprefix -I, $(VERILOG_INCLUDE_DIRS))
endif
```

I have verifiyed them all locally and the appear to correct correctly when VERILOG_INCLUDE_DIRS  is present. And the `COMPILE_ARGS +=` way also still works. 

I do not have access to vcs to check this feature, so I have not added it, but the way one can enhance them should be obvious

I am opening an issue to help track, but the bulk of the work is done already in a branch on my fork repo. I am unsure how to add tests for this addition or the news fragments to add this to.

[Daxzio verilog_include](https://github.com/daxzio/cocotb/tree/verilog_include)

forensicgarlic, 2023-01-12:
similarly for verilator I've done 
```
ifdef VERILOG_INCLUDE_DIRS
    VERILATOR_INCLUDE_DIRS=$(foreach d, ${VERILOG_INCLUDE_DIRS}, -I$d)
endif
```
and added the `$VERILATOR_INCLUDE_DIRS` to the call to verilator.

themperek, 2023-01-18:
I don't know if we will add any new features to Makefiles anymore. There are a few such features in [python runner](https://github.com/cocotb/cocotb/blob/master/cocotb/runner.py)

davekeeshan, 2023-01-18:
@imphil 

I have added a PR, #3189, with the updates I have.  I only have access to icarus, verilator, ius and xcelium to verify (which is the most I have ever had access to in my 25 year career, usually you only have one).  

You can merge it or not, or maybe it can be used as a based to fill out the remaining verilog based simulators.

However, it is a constructive addition, it doesn't break anything that exists already, and there already is support for `VERILOG_INCLUDES_DIRS `in questa/modelsim, so the current repo is in partially implemented state right now.

davekeeshan, 2023-01-18:
@themperek 

I would like to point out that all the reference documentation only talk about Makefiles:

https://docs.cocotb.org/en/stable/quickstart.html

Other people I have heard are using the python runner, but it is almost like a myth on the wind.  When  I go looking for examples ...

I suggest you either update your quickstart and only detail the python runner or try and keep the Makefiles in a basic level of sync (I view VERILOG_INCLUDE_DIRS in the basic level catoegory).  

I know, as an industry veteran, that the level of experience of users are actually quite low, they struggle to understand basic Makefile syntax, I am currently advocating how clean cocotb is, but something as simple as this is enough for them to pass it by (its very fustrating, that probably why tcl is still the dominant language in eda tools, grrr!)

themperek, 2023-01-18:
@davekeeshan Python runner is a very new thing not been released yet. We will be slowly moving. Makefiles will stay.

davekeeshan, 2023-01-18:
@themperek 

Jebus, then it will be like python2, you'll have to support makefiles for 10 years, at least ;)

Seriously though, if it is not released yet, not is not fair to act like it is gone, people will be trading off internet examples for years, that is how it is done.

forensicgarlic, 2023-01-20:
thanks for making the pull-request @davekeeshan

PR #3189 (2023-01-26): Add VERILOG_INCLUDE_DIRS to Makefiles
Support in the makefiles for icarus, ius, xcelium, questa, riviera, vcs, and verilator. Closes #3185.
closed 2023-01-26 by imphil