cocotb/cocotb #3186: Improve support for assertion rewriting regardles of test name

TrampHRC, 2023-01-13:
The pull request #3011 added this support.

The following lines, written by @ktbarrett, force assertion rewriting of python modules listed in the `MODULE` env variable :
https://github.com/cocotb/cocotb/blob/4b5b80eb409c6f7e2a816784fe1086c785fb928f/cocotb/regression.py#L255-L258

This has 2 issues:

- It doesn't handle the case of `__init__.py` packages.
- It doesn't handle modules in packages : `pkg.module` => `pkg.module.py` which does not throw an error but does not allow for assertion rewriting.

TrampHRC, 2023-01-13:
@ktbarrett Out of curiosity, what would be the downside of assertion rewriting all files : `-o python_files=*.py` ?

ktbarrett, 2023-01-13:
asserts that have nothing to do with test failure (e.g. one's buried in support library code) would be rewritten. I guess such asserts are currently considered valid test-ending exceptions; meaning test checking and internal self checking are already munged together (yuck) so who cares if the assertion rewriting is not differentiated. You've convinced me, rewriting all files would be just fine by me.

ktbarrett, 2023-01-19:
@TrampHRC If you want to open a PR which changes the logic of the assertion rewriting I will review it. As discussed you can just change the pattern to `*.py`.

PR #3223 (2023-02-03): Allow assertion rewriting in all Python source files
Closes #3186.
closed 2023-02-03 by ktbarrett