cocotb/cocotb #3190: Discovering VHPI objects in Riviera Pro behaves incorrectly

jwprice100, 2023-01-19:
### Context
Cocotb version: 1.7.2 (and [latest](https://github.com/cocotb/cocotb/commit/b48ef1f67fb7ccf1a525fbe7fd417d8021738b51))
Operating System: Ubuntu 22.04 64 bit
Simulator: Riviera Pro 2022.10 64 bit
Python: 3.6 system and 3.11 built from source
Language: VHDL/VHPI

### Description
cocotb is not able to discover all objects in Riviera Pro. It seems when iterating over the various regions, if it encounters a logic vector type, the iteration for that region terminates early. For example when iterating over ports, as soon as a port with type std_logic_vector is encountered, that port has an objected created but then the iteration stops.

Note that these undiscovered objects can be accessed directly. Furthermore accessing them adds to them the attributes of the HierarchyObject. The only work around I've failed is to access all desired signals first. 

Another note, I do not see this behavior when using VPI.

### How to Reproduce
Run the attached example in Riviera Pro. The test will fail when it checks to see if port D exists. This is because port C is first and it is a std_logic_vector. I've also attached some log output with COCOTB_LOG_LEVEL set to TRACE. 
[log.txt](https://github.com/cocotb/cocotb/files/10461609/log.txt)
[riviera_discovery_test.zip](https://github.com/cocotb/cocotb/files/10461580/riviera_discovery_test.zip)

marlonjames, 2023-05-22:
There is a problem with `get_range()`: #3312 
The fix may solve this issue.

imphil, 2023-05-22:
There is also https://github.com/cocotb/cocotb/pull/3310, which goes into the same direction.

mdx83, 2023-05-23:
I confirm that #3310 also fixes this issue.

closed 2023-05-23 by imphil