cocotb/cocotb #3194: [tracking] Get the cocotb test suite passing with Verilator

imphil, 2023-01-22:
This issue tracks progress towards getting the cocotb test suite passing with Verilator.

Status updates:
- [x] Timed callbacks. Fixed in Verilator 5.004 (https://github.com/verilator/verilator/commit/c2bdd06fccd4ef551948ed6dfe0eff76004fa117), so we're back to where we roughly were with Verilator 4.106.
- [ ] Once we have a passing CI suite: Update documentation to indicate which Verilator version is supported.

Various other test failures need to be diagnosed and issues have to be created.

imphil, 2023-01-22:
Test failures with Verilator master as of today:

```
Failure in testsuite: 'all' classname: 'test_iteration_es' testcase: 'recursive_discovery' with parameters 'all'
Error: Test test_iteration_es:recursive_discovery failed
Failure in testsuite: 'all' classname: 'test_discovery' testcase: 'pseudo_region_access' with parameters 'all'
Error: Test test_discovery:pseudo_region_access failed
Failure in testsuite: 'all' classname: 'test_discovery' testcase: 'access_signal' with parameters 'all'
Error: Test test_discovery:access_signal failed
Failure in testsuite: 'all' classname: 'test_discovery' testcase: 'access_single_bit' with parameters 'all'
Error: Test test_discovery:access_single_bit failed
Failure in testsuite: 'all' classname: 'test_discovery' testcase: 'access_integer' with parameters 'all'
Error: Test test_discovery:access_integer failed
Failure in testsuite: 'all' classname: 'test_discovery' testcase: 'access_const_string_verilog' with parameters 'all'
Error: Test test_discovery:access_const_string_verilog failed
Failure in testsuite: 'all' classname: 'test_discovery' testcase: 'access_var_string_verilog' with parameters 'all'
Error: Test test_discovery:access_var_string_verilog failed
Failure in testsuite: 'all' classname: 'test_discovery' testcase: 'access_internal_register_array' with parameters 'all'
Error: Test test_discovery:access_internal_register_array failed
Failure in testsuite: 'all' classname: 'test_discovery' testcase: 'access_gate' with parameters 'all'
Error: Test test_discovery:access_gate failed
Failure in testsuite: 'all' classname: 'test_discovery' testcase: 'type_check_verilog' with parameters 'all'
Error: Test test_discovery:type_check_verilog failed
Failure in testsuite: 'all' classname: 'issue_330' testcase: 'issue_330_direct' with parameters 'all'
Error: Test issue_330:issue_330_direct failed
Failure in testsuite: 'all' classname: 'issue_330' testcase: 'issue_330_iteration' with parameters 'all'
Error: Test issue_330:issue_330_iteration failed
Failure in testsuite: 'all' classname: 'test_cocotb_array' testcase: 'test_in_vect_unpacked' with parameters 'all'
Error: Test test_cocotb_array:test_in_vect_unpacked failed
Failure in testsuite: 'all' classname: 'test_cocotb_array' testcase: 'test_in_2d_vect_unpacked_unpacked' with parameters 'all'
Error: Test test_cocotb_array:test_in_2d_vect_unpacked_unpacked failed
Failure in testsuite: 'all' classname: 'test_cocotb_array' testcase: 'test_in_vect_packed_packed_unpacked' with parameters 'all'
Error: Test test_cocotb_array:test_in_vect_packed_packed_unpacked failed
Failure in testsuite: 'all' classname: 'test_cocotb_array' testcase: 'test_in_vect_packed_unpacked_unpacked' with parameters 'all'
Error: Test test_cocotb_array:test_in_vect_packed_unpacked_unpacked failed
Failure in testsuite: 'all' classname: 'test_cocotb_array' testcase: 'test_in_vect_unpacked_unpacked_unpacked' with parameters 'all'
Error: Test test_cocotb_array:test_in_vect_unpacked_unpacked_unpacked failed
Failure in testsuite: 'all' classname: 'test_cocotb_array' testcase: 'test_in_arr_packed_unpacked' with parameters 'all'
Error: Test test_cocotb_array:test_in_arr_packed_unpacked failed
Failure in testsuite: 'all' classname: 'test_cocotb_array' testcase: 'test_in_arr_unpacked_unpacked' with parameters 'all'
Error: Test test_cocotb_array:test_in_arr_unpacked_unpacked failed
Failure in testsuite: 'all' classname: 'test_cocotb_array' testcase: 'test_in_2d_arr_unpacked' with parameters 'all'
Error: Test test_cocotb_array:test_in_2d_arr_unpacked failed
Failure in testsuite: 'all' classname: 'test_force_release' testcase: 'test_force_release' with parameters 'all'
Error: Test test_force_release:test_force_release failed
Failure in testsuite: 'all' classname: 'test_long_log_msg' testcase: 'test_access_long_name' with parameters 'all'
Error: Test test_long_log_msg:test_access_long_name failed
Failure in testsuite: 'all' classname: 'test_array_simple' testcase: 'test_ndim_array_handles' with parameters 'all'
Error: Test test_array_simple:test_ndim_array_handles failed
Failure in testsuite: 'all' classname: 'test_array_simple' testcase: 'test_ndim_array_indexes' with parameters 'all'
Error: Test test_array_simple:test_ndim_array_indexes failed
Failure in testsuite: 'all' classname: 'test_array_simple' testcase: 'test_struct' with parameters 'all'
Error: Test test_array_simple:test_struct failed
```

Run: https://github.com/cocotb/cocotb/actions/runs/3981649305/jobs/6825520210

marlonjames, 2023-01-23:
I have an old branch where I started going through the tests. I'll give it a look soon.

forensicgarlic, 2023-02-05:
https://github.com/cocotb/cocotb/pull/3231 is currently trying to update the Makefile (and suggested edit for the documentation)

PR #3740 (2024-03-03): Docs: Update Verilator support section
Closes #3194
Add note suggesting to try the newest release if there users encounter issues.
Note the `--timing` option for delayed assignment support.
closed 2024-03-03 by marlonjames