cocotb/cocotb #3196: Regression manager does not handle failed pytest.raises() result

marlonjames, 2023-01-24:
cococtb: master (4aee0a0895a36606f9263df9370405663f496d22)
Ubuntu 20.04.2 LTS on WSL2
simulator: Verilator 5.007 devel rev v5.006-7-g4a8cfe367

When `pytest.raises()` fails, it raises a `Failed` exception.

This exception derives from `BaseException`, so the check in `RegressionManager._score_test()` fails:
https://github.com/cocotb/cocotb/blob/4aee0a0895a36606f9263df9370405663f496d22/cocotb/regression.py#L395-L400

 The raised `Failed` exception propagates up through `Scheduler._test_completed()` and into the simulator extension module.

Here is an example traceback:
```
567237.42ns INFO     cocotb.regression                  running test_integer_overflow (158/183)
                                                          Test integer overflow.
ERROR: called callback function threw exception
Traceback (most recent call last):
  File "/home/marlon/git/cocotb/.nox/dev_test_sim-sim-verilator-toplevel_lang-verilog-gpi_interface-vpi/lib/python3.8/site-packages/cocotb/scheduler.py", line 349, in _test_completed
    self._handle_result(test)
  File "/home/marlon/git/cocotb/.nox/dev_test_sim-sim-verilator-toplevel_lang-verilog-gpi_interface-vpi/lib/python3.8/site-packages/cocotb/regression.py", line 332, in _handle_result
    self._record_result(
  File "/home/marlon/git/cocotb/.nox/dev_test_sim-sim-verilator-toplevel_lang-verilog-gpi_interface-vpi/lib/python3.8/site-packages/cocotb/regression.py", line 506, in _record_result
    test_pass, sim_failed = self._score_test(test, outcome)
  File "/home/marlon/git/cocotb/.nox/dev_test_sim-sim-verilator-toplevel_lang-verilog-gpi_interface-vpi/lib/python3.8/site-packages/cocotb/regression.py", line 396, in _score_test
    outcome.get()
  File "/home/marlon/git/cocotb/.nox/dev_test_sim-sim-verilator-toplevel_lang-verilog-gpi_interface-vpi/lib/python3.8/site-packages/cocotb/outcomes.py", line 53, in get
    raise self.error
  File "/home/marlon/git/cocotb/tests/test_cases/test_cocotb/test_handle.py", line 312, in test_integer_overflow
    await int_overflow_test(dut.stream_in_int, 32, "ovfl", limits)
  File "/home/marlon/git/cocotb/tests/test_cases/test_cocotb/test_handle.py", line 148, in int_overflow_test
    signal.value = value
  File "/home/marlon/git/cocotb/.nox/dev_test_sim-sim-verilator-toplevel_lang-verilog-gpi_interface-vpi/lib/python3.8/site-packages/_pytest/python_api.py", line 984, in __exit__
    fail(self.message)
  File "/home/marlon/git/cocotb/.nox/dev_test_sim-sim-verilator-toplevel_lang-verilog-gpi_interface-vpi/lib/python3.8/site-packages/_pytest/outcomes.py", line 194, in fail
    raise Failed(msg=reason, pytrace=pytrace)
Failed: DID NOT RAISE <class 'OverflowError'>
- :0: Verilog $finish
- /home/marlon/git/cocotb/.nox/dev_test_sim-sim-verilator-toplevel_lang-verilog-gpi_interface-vpi/lib/python3.8/site-packages/cocotb/share/lib/verilator/verilator.cpp:118: Verilog $finish
- /home/marlon/git/cocotb/.nox/dev_test_sim-sim-verilator-toplevel_lang-verilog-gpi_interface-vpi/lib/python3.8/site-packages/cocotb/share/lib/verilator/verilator.cpp:118: Second verilog $finish, exiting
ERROR: results.xml was not written by the simulation!
```

ktbarrett, 2023-01-25:
>This exception derives from BaseException

Yikers... Do we treat `Failure` as a test failure or a test exception? We might need to special case it here.

ktbarrett, 2023-02-03:
`Failed` is only available as a private implementation detail (`_pytest.outcomes.Failed`). I'm not sure how often the internals change, but the whole thing is starting to stink. I have a clever way of getting around it though, but I don't like it.

PR #3227 (2023-02-13): Catch exceptions derived from BaseException inside Tasks
Closes #3196.
Alternative to #3226.

Exceptions that are raised inside coroutines running in Tasks are captured inside an outcomes.Error object. Previously, when this object was unwrapped in the Scheduler or RegressionManager, the exception inside was not handled correctly if the exception was derived from BaseException and not Exception.

This was the case with pytest's Failed exception that is raised when the following checks fail in a Task:
* pytest.raises
* pytest.warns
* pytest.deprecated_call