cocotb/cocotb #3197: But in cocotb.regression.py _my_import. Not working correctly on Windows

3phreal, 2023-01-25:
CocoTb version 1.7.2
Windows/Linux Mint 64 Bit
Riviera-pro 2020.10.120.7919
Python version 3.7 self-installed

I have just started using CocoTb so I do not have the grand overview yet. However, I have been trying to get it to run on Windows. 

Since I run simulations on both windows and Linux systems, having this option is a requirement, since I already generate my own .tcl scripts. Including the extra lines from CocoTb's make system into ours where reasonably easy. 

I duplicated the setup on our windows system and it was up and running also. However, I had problems including my Test Module defined by the environment variable module. 

I can see that CocoTb regression uses the old __import__() to import said modules. And what I found was that on Linux I could give the full path and module name and it could import it without any issues. 
However, on windows, I had to change my CWD to the folder location of the module and then import it for this to work. 

I changed the regression.py:

```
def _my_import(name: str) -> Any:
    mod = __import__(name)
    components = name.split(".")
    for comp in components[1:]:
        mod = getattr(mod, comp)
    return mod
```

to 

```
def _my_import(name: str) -> Any:
    current = os.getcwd()
    os.chdir(os.path.dirname(name))
    fmtname = os.path.basename(name)
    mod = __import__(fmtname)
    components = fmtname.split(".")
    for comp in components[1:]:
        mod = getattr(mod, comp)
    os.chdir(current)
    return mod
```

This is not the prettiest change, but it works right now. 
I understand that the usage of __import__() is discouraged, or is there something I've overlooked?

But the above change enables me to run the entire thing on Windows without using make or WSL. 

Regards

ktbarrett, 2023-01-25:
Sounds like you are not setting PYTHONPATH relative to where your test modules are defined? We can change `__import__()` to `importlib.import_module`, though I don't see how that will help you.

3phreal, 2023-01-25:
I do not set PYTHONPATH one either. However, doing so on Windows fixes the problem.