cocotb/cocotb #3206: Signals not getting set in verilator

alexforencich, 2023-01-28:
I know that verilator support is still a bit, shall we say, experimental, but since they just released 5.006 with ostensibly fixes some of the major issues that were blocking cocotb from working, I decided to give it a shot and see what things looked like.

My initial testing with running the tests in cocotbext-axi resulted in some very odd behavior: it seems like all of the signals are stuck at 0, except for the clock.  Any `dut.some_signal.value = something` seem to have no effect at all.  Even `setimmediatevalue()` seems to have no effect, the waveform dump always shows everything as 0, with one exception - any clocks driven by `cocotb.clock.Clock` appear to be working as expected.  Now, since `cocotb.clock.Clock` seems to drive signals in exactly the same way, I am really at a loss as to what might be going on here.

I have added print statements here and there to verify that all of the assignments are actually getting hit.  I also tried using the older assignment style with `<=`, with the same results.  On the "stub" module that I am testing with, I tried changing all of the ports from "inout" to "input" and I have also tried timescales of "1ns/1ns" and "1ns/1ps", again with the same result.

All of this is with verilator 5.006 and the latest git version of cocotb.

Is anyone else who is experimenting with verilator seeing similar behavior?  Or even different behavior?  Is there anything else that I can do to debug what is going on between cocotb and verilator?

alexforencich, 2023-01-28:
Ah, after spending quite a bit of time commenting things out across several different python packages, I think I might have found the culprit: `dir`.  It gets called in cocotb-bus in `_caseInsensGetattr` when creating the `Bus` object, if `case_insensitive` is not set to `False`.  So, something about `dir` getting called seems to be screwing something up somewhere, not sure if this is a cocotb problem or a verilator problem.

After neutering `_caseInsensGetattr` so it doesn't call `dir`, all of the testbenches for cocotbext-axi appear to run correctly with verilator.

marlonjames, 2023-01-28:
Same as #2473.
Root cause is detailed in #2244.

closed 2023-01-28 by marlonjames