cocotb/cocotb #3209: System cannot find the file specified (Makefile.sim)

AhsanAliUet, 2023-01-29:
I have installed cocotb using pip and all my environment variables are in correct position. When I try to run the command ```make SIM=icarus``` I get the error ```System cannot find the file specified```. Attached is a picture for reference. How can I solve this problem?

[image]

themperek, 2023-01-29:
@AhsanAliUet Using Makefiles on Widows is very "tricky" and hard to help. One workaround would be use python to run test instead of Makefiles.

AhsanAliUet, 2023-01-29:
How can I use python to simulate a Verilog or System Verilog code in some simulator? I mean, I can see the python test benches in in repository, but they all use Makefiles. How can I avoid Makefiles while using python?

themperek, 2023-01-29:
Some options:
- cocotb python runner (need master) https://docs.cocotb.org/en/latest/library_reference.html
- cocotb-test https://github.com/themperek/cocotb-test
- https://github.com/olofk/fusesocotb
- ...

AhsanAliUet, 2023-01-29:
@themperek Thanks for your support. Really mean it.

closed 2023-01-29 by AhsanAliUet