cocotb/cocotb #3215: Bug in regression manager `_trim` function on empty docstrings

ktbarrett, 2023-01-30:
When the docstring is empty (i.e. `""""""`) the index in the following line causes an `IndexError`.

https://github.com/cocotb/cocotb/blob/3aa6777dd3d0ff221e6f6b4de70051fe20a7e414/cocotb/regression.py#L912

marlonjames, 2023-01-30:
Looks like the change from
```py3
if not docstring:
  return ''
```
to
```py3
if docstring is None:
  return ""
```
missed the case for an empty string.

ktbarrett, 2023-01-30:
Hmmm... I'm guessing this was done to appease the type checker (the argument is `Optional`) without realizing it was supposed to cover both `None` and `""`. That should be made more explicitly:

```python
if docstring is None or docstring == "":
  return ""
```

PR #3220 (2023-02-03): Fix issue with _trim on empty docstrings
Closes #3215.
closed 2023-02-03 by ktbarrett