cocotb/cocotb #3219: Unexpected Behaviour in Coroutine Termination

aqeelmahroof, 2023-02-03:
I encountered an issue while running a test with a reset aware mechanism. The coroutines are started by a runner task which monitors for reset and kills the executing coroutines.
```
async def runner_with_reset(rst,func):
    seq_coro = cocotb.start_soon(func())
    log.info("Runner::started " + func.__name__)
    await First(seq_coro, rst.wait())
    log.info("Runner::triggerd " + func.__name__)
    if rst.is_set():
        #if the reset s trigered kill the coroutine
        seq_coro.kill()
        log.info("Runner::killed " + func.__name__)
    else:
        log.info("Runner::finished " + func.__name__)
```
In the test case `test_multi_func_reset`, the `top_body` coroutine is started via the runner, and `top_body starts` child_body also via the runner. The expected behaviour is that when the reset is triggered, both `top_body` and `child_body` should be terminated. However, only `top_body` is terminated and the results are unexpected.

I have provided a simple test case, for your review. The two unexpected behaviours are:
1. The child_body coroutine is not terminated
2. The await First(seq_coro, rst.wait()) did not get triggered while the runner was executing for child_body

Thank you for your attention to this matter. I look forward to your response.

TestCase:
[reset_coro_error.zip](https://github.com/cocotb/cocotb/files/10574405/reset_coro_error.zip)

Environment: 
* Cocotb version v1.7.2
* Windows 64 bit
* Questa 64bit
* Python 3.9.7 Miniconda

marlonjames, 2023-02-03:
The problem is that the Task running your `top_body` coroutine is executing `runner_with_reset` monitoring the separate Task running `child_body`.
When the `Event` triggers is set, the Task running `top_body` is killed before it responds to the same `Event`. This makes `child_body` an orphan Task, and it completes without being killed.

Loosely, these are the parallel tasks and the coroutine call stacks:
```
-> Test test_multi_func_reset
-> Task 2
      runner_with_reset(reset_event, top_body)
-> Task 3
      top_body()
      runner_with_reset(reset_event, child_body)
-> Task 4
     child_body()
```
When the `Event` is set, Task 2 runs first, which kills Task 3. Task 3 was waiting on the same `Event`, but it was killed so it doesn't run, and so Task 4 is not killed.

cocotb does not guarantee the order that Tasks run if multiple Tasks are waiting on the same trigger.

aqeelmahroof, 2023-02-03:
Thank you, @marlonjames, for providing clarification. Your explanation aligns with what I have observed in the logs.

closed 2023-02-03 by aqeelmahroof