cocotb/cocotb #3229: Performance regression in cocotb 1.7

ktbarrett, 2023-02-03:
I had an engineer report a significant performance regression when upgrading from 1.6.2 to 1.7.2. This is the not the first time this complaint has been heard, there was someone on the Gitter who mentioned it as well (it's blocked for me rn so I can't go find it).

marlonjames, 2023-02-03:
This might be the same as #3111

ktbarrett, 2023-02-03:
I'm running a bisect now, it may take a while. Apparently the performance decreases as the run continues, pointing to a leak.

ktbarrett, 2023-02-06:
The culprit is f6d5c329e3cacaf0af4fbcfa3f8c5e11834b2665. So this only affects Questa. I am somewhat uncertain as to why.

marlonjames, 2023-02-06:
Looks like it's because `FliTimedCbHdl::cleanup_callback()` sets the state to `GPI_DELETE` before calling `FliProcessCbHdl::cleanup_callback()`.
https://github.com/cocotb/cocotb/blob/918fc4118d5b71ec6f5a5eba89a235d995de6d56/cocotb/share/lib/fli/FliCbHdl.cpp#L73-L92

So `mti_Desensitize()` is no longer called.
https://github.com/cocotb/cocotb/blob/918fc4118d5b71ec6f5a5eba89a235d995de6d56/cocotb/share/lib/fli/FliCbHdl.cpp#L43-L49

ktbarrett, 2023-02-06:
>              /* Issue #188: Work around for modelsim that is harmless to othes 
>                too, we tag the time as delete, let it fire then do not pass up 
>                */ 

What you're saying is that this workaround is no longer valid, or maybe was only valid for 10.3 where calling desensitize was broken, but now shouldn't be done because of performance issues?

marlonjames, 2023-02-07:
I'm not sure how that change could cause a slow down.
The FLI documentation says that only one scheduled wakeup can exist at a time for a process.

> `mti_ScheduleWakeup()` schedules the specified process to be called after the specified delay. A process can have no more than one pending wake-up call. A call to `mti_ScheduleWakeup()` cancels a prior pending wake-up call for the specified process regardless of the delay values. 

So even without calling `mti_Desensitize()`, the next time that Timed callback is used, it should cancel any already-scheduled wake-up.

---

I don't know if issue #188 is still a problem with later versions of Modelsim.

I also don't think the problem is related to FLI. The workaround was added first for VPI (#189), then brought over to FLI.
It's most likely some issue with how Modelsim is tracking callback objects and handles.

From #188:
> There seems to be an issue with modelsim 10.3x where de-registering a Timer callback in the same timestep. No error is reported by modelsim but later on it during some internal processing it segfaults.

The workaround that was implemented is to never deregister the callback,
but instead cocotb waits for it to fire and ignores it.
This causes issues such as #2969 for `Timer` triggers far into the future.

ktbarrett, 2023-02-07:
> I'm not sure how that change could cause a slow down.

That was my take, I could add a trace to see how many times the callbacks are called to see if there is some weird leak because they aren't being deregistered. Perhaps they have been changed to multi-shot callbacks?

> This causes issues such as https://github.com/cocotb/cocotb/issues/2969 for Timer triggers far into the future.

Nice catch!

ktbarrett, 2023-02-14:
Well trying to run `mti_Desensitize` on Timer objects like we do in the VPI causes a stacktrace. Yay, more testing...

PR #3248 (2023-02-16): Correctly desensitize timer callbacks in FLI
Small test shows 5x FLI performance. Closes #3229. The whole callback cleanup needs some love, but this is the minimal change to fix the performance regression.
closed 2023-02-16 by ktbarrett