cocotb/cocotb #3230: 17 errors in tests

yurivict, 2023-02-04:
```
========================================================================================== test session starts ==========================================================================================
platform freebsd13 -- Python 3.9.16, pytest-7.2.0, pluggy-1.0.0 -- /usr/local/bin/python3.9
cachedir: .pytest_cache
hypothesis profile 'default' -> database=DirectoryBasedExampleDatabase('/usr/ports/cad/py-cocotb/work-py39/cocotb-1.7.2/tests/.hypothesis/examples')
Using --randomly-seed=1755101744
rootdir: /usr/ports/cad/py-cocotb/work-py39/cocotb-1.7.2, configfile: setup.cfg
plugins: hypothesis-6.58.0, asyncio-0.20.3, trio-0.8.0, cov-2.9.0, flaky-3.7.0, xdist-2.5.0, flake8-1.1.1, forked-1.4.0, anyio-3.6.2, randomly-3.12.0, timeout-2.1.0, rerunfailures-10.1, env-0.6.2, freezegun-0.4.2, mock-1.10.4, nbval-0.10.0
asyncio: mode=strict
collected 217 items / 17 errors                                                                                                                                                                         

================================================================================================ ERRORS =================================================================================================
______________________________________________________________________ ERROR collecting tests/test_cases/test_array/test_array.py _______________________________________________________________________
import file mismatch:
imported module 'test_array' has this __file__ attribute:
  /usr/ports/cad/py-cocotb/work-py39/cocotb-1.7.2/tests/pytest/test_array.py
which is not the same as the test file we want to collect:
  /usr/ports/cad/py-cocotb/work-py39/cocotb-1.7.2/tests/test_cases/test_array/test_array.py
HINT: remove __pycache__ / .pyc files and/or use a unique basename for your test file modules
_______________________________________________________________ ERROR collecting tests/test_cases/test_array_simple/test_array_simple.py ________________________________________________________________
test_cases/test_array_simple/test_array_simple.py:22: in <module>
    expect_error=Exception if cocotb.SIM_NAME.lower().startswith("ghdl") else ()
E   AttributeError: 'NoneType' object has no attribute 'lower'
___________________________________________________________________ ERROR collecting tests/test_cases/test_cocotb/test_deprecated.py ____________________________________________________________________
test_cases/test_cocotb/test_deprecated.py:31: in <module>
    if cocotb.SIM_NAME.lower().startswith("icarus")
E   AttributeError: 'NoneType' object has no attribute 'lower'
_____________________________________________________________________ ERROR collecting tests/test_cases/test_cocotb/test_handle.py ______________________________________________________________________
test_cases/test_cocotb/test_handle.py:18: in <module>
    SIM_NAME = cocotb.SIM_NAME.lower()
E   AttributeError: 'NoneType' object has no attribute 'lower'
____________________________________________________________________ ERROR collecting tests/test_cases/test_cocotb/test_scheduler.py ____________________________________________________________________
test_cases/test_cocotb/test_scheduler.py:330: in <module>
    expect_error=Exception if cocotb.SIM_NAME.lower().startswith("ghdl") else ()
E   AttributeError: 'NoneType' object has no attribute 'lower'
_________________________________________________________________ ERROR collecting tests/test_cases/test_cocotb/test_timing_triggers.py _________________________________________________________________
test_cases/test_cocotb/test_timing_triggers.py:133: in <module>
    expect_fail=cocotb.SIM_NAME.lower().startswith(("icarus", "ncsim", "xmsim")),
E   AttributeError: 'NoneType' object has no attribute 'lower'
______________________________________________________________________ ERROR collecting tests/test_cases/test_cocotb/test_utils.py ______________________________________________________________________
import file mismatch:
imported module 'test_utils' has this __file__ attribute:
  /usr/ports/cad/py-cocotb/work-py39/cocotb-1.7.2/tests/pytest/test_utils.py
which is not the same as the test file we want to collect:
  /usr/ports/cad/py-cocotb/work-py39/cocotb-1.7.2/tests/test_cases/test_cocotb/test_utils.py
HINT: remove __pycache__ / .pyc files and/or use a unique basename for your test file modules
__________________________________________________________________ ERROR collecting tests/test_cases/test_discovery/test_discovery.py ___________________________________________________________________
test_cases/test_discovery/test_discovery.py:39: in <module>
    expect_error=IndexError if cocotb.SIM_NAME.lower().startswith("ghdl") else ()
E   AttributeError: 'NoneType' object has no attribute 'lower'
__________________________________________________________________ ERROR collecting tests/test_cases/test_discovery/test_vhdl_block.py __________________________________________________________________
test_cases/test_discovery/test_vhdl_block.py:33: in <module>
    expect_error=AttributeError if cocotb.SIM_NAME.lower().startswith("ghdl") else ()
E   AttributeError: 'NoneType' object has no attribute 'lower'
____________________________________________________________________ ERROR collecting tests/test_cases/test_exit_error/test_exit.py _____________________________________________________________________
test_cases/test_exit_error/test_exit.py:12: in <module>
    NameErrorLol  # noqa: F821
E   NameError: name 'NameErrorLol' is not defined
___________________________________________________________________ ERROR collecting tests/test_cases/test_external/test_external.py ____________________________________________________________________
test_cases/test_external/test_external.py:93: in <module>
    if cocotb.SIM_NAME.startswith(("xmsim", "ncsim")) and cocotb.LANGUAGE in ["vhdl"]
E   AttributeError: 'NoneType' object has no attribute 'startswith'
______________________________________________________________________ ERROR collecting tests/test_cases/test_fatal/test_fatal.py _______________________________________________________________________
test_cases/test_fatal/test_fatal.py:12: in <module>
    expect_error=SimFailure, skip=cocotb.SIM_NAME.lower().startswith("riviera")
E   AttributeError: 'NoneType' object has no attribute 'lower'
______________________________________________________________ ERROR collecting tests/test_cases/test_force_release/test_force_release.py _______________________________________________________________
test_cases/test_force_release/test_force_release.py:8: in <module>
    @cocotb.test(expect_fail=cocotb.SIM_NAME.lower().startswith("ghdl"))
E   AttributeError: 'NoneType' object has no attribute 'lower'
[... 26 lines cut ...]

pytest/test_parallel_cocotb.py:24
  /usr/ports/cad/py-cocotb/work-py39/cocotb-1.7.2/tests/pytest/test_parallel_cocotb.py:24: PytestUnknownMarkWarning: Unknown pytest.mark.compile - is this a typo?  You can register custom marks to avoid this warning - for details, see https://docs.pytest.org/en/stable/how-to/mark.html
    @pytest.mark.compile

test_cases/test_cocotb/test_testfactory.py:45
  /usr/ports/cad/py-cocotb/work-py39/cocotb-1.7.2/tests/test_cases/test_cocotb/test_testfactory.py:45: PytestCollectionWarning: cannot collect test class 'TestClass' because it has a __init__ constructor (from: tests/test_cases/test_cocotb/test_testfactory.py)
    class TestClass(Coroutine):

test_cases/test_compare/test_compare.py:14
  /usr/ports/cad/py-cocotb/work-py39/cocotb-1.7.2/tests/test_cases/test_compare/test_compare.py:14: PytestCollectionWarning: cannot collect test class 'Testbench' because it has a __init__ constructor (from: tests/test_cases/test_compare/test_compare.py)
    class Testbench:

../../../../../../local/lib/python3.9/site-packages/pytest_freezegun.py:17: 434 warnings
  /usr/local/lib/python3.9/site-packages/pytest_freezegun.py:17: DeprecationWarning: distutils Version classes are deprecated. Use packaging.version instead.
    if LooseVersion(pytest.__version__) < LooseVersion('3.6.0'):

-- Docs: https://docs.pytest.org/en/stable/how-to/capture-warnings.html
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!! Interrupted: 17 errors during collection !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
=================================================================================== 438 warnings, 17 errors in 2.54s ====================================================================================
```

Version: 1.7.2
Python-3.9
FreeBSD 13.1

imphil, 2023-02-04:
Well, SIM_NAME seems to be not set. How are you running your test? What simulator are you running against?

themperek, 2023-02-04:
You can use `nox` for example `nox -s "dev_test_sim(sim='icarus', toplevel_lang='verilog', gpi_interface='vpi')"`. This will set all variables.

yurivict, 2023-02-04:
Adding the environment variable ```SIM_NAME=iverilog``` doesn't make any difference.

What's "nox"?

themperek, 2023-02-04:
> Adding the environment variable `SIM_NAME=iverilog` doesn't make any difference.

`SIM=icarus`

yurivict, 2023-02-04:
```SIM=icarus```also doesn't make a difference.

It doesn't initialize ```cocotb.SIM_NAME```. The tests are run with PyTest.

marlonjames, 2023-02-04:
cocotb's `setup.cfg` should restrict `pytest` from finding tests in `tests/test_cases`:

https://github.com/cocotb/cocotb/blob/918fc4118d5b71ec6f5a5eba89a235d995de6d56/setup.cfg#L19-L24

imphil, 2023-02-04:
The question remains: how do you run the tests? Can you post the build and test commands that lead to this failure?

yurivict, 2023-02-05:
[Here](https://people.freebsd.org/~yuri/py39-cocotb-1.7.2.log) is a complete log with all commands.

marlonjames, 2023-02-05:
I've removed some of the environment variables in the command line to show the relevant parts from your log:
`cd /usr/ports/cad/py-cocotb/work-py39/cocotb-1.7.2/tests && PYTHONPATH=/usr/ports/cad/py-cocotb/work-py39/stage/usr/local/lib/python3.9/site-packages  SIM=icarus /usr/local/bin/python3.9 -m pytest -k '' -rs -v -o addopts=`

The `pytest` tests must be run from the root directory, not from `tests`.
These are also only a subset of the cocotb testsuite.
Running `make test` will run the cocotb testsuite with a simulator.
You can also use the `nox` Python package to run everything together.

yurivict, 2023-02-05:
I can run tests now with ```make test```, thank you.

I will create a separate bug report for test failures.

closed 2023-02-05 by yurivict