cocotb/cocotb #3236: Manuy tests in the testsuite fail: pytest.PytestDeprecationWarning: The hookimpl pytest_load_initial_conftests uses old-style configuration options

yurivict, 2023-02-05:
```
/usr/local/bin/iverilog -o sim_build/sim.vvp -D COCOTB_SIM=1 -s endian_swapper_sv -f sim_build/cmds.f -g2012   endian_swapper.sv
rm -f results.xml
MODULE=test_iteration_es TESTCASE= TOPLEVEL=endian_swapper_sv TOPLEVEL_LANG=verilog \
         /usr/local/bin/vvp -M /usr/ports/cad/py-cocotb/work-py39/stage/usr/local/lib/python3.9/site-packages/cocotb/libs -m libcocotbvpi_icarus   sim_build/sim.vvp 
     -.--ns INFO     gpi                                ..mbed/gpi_embed.cpp:78   in set_program_name_in_venv        Did not detect Python virtual environment. Using system-wide Python interpreter
     -.--ns INFO     gpi                                ../gpi/GpiCommon.cpp:101  in gpi_print_registered_impl       VPI registered
     0.00ns INFO     cocotb                             Running on Icarus Verilog version 11.0 (stable)
     0.00ns INFO     cocotb                             Running tests with cocotb v1.7.2 from /usr/ports/cad/py-cocotb/work-py39/stage/usr/local/lib/python3.9/site-packages/cocotb
     0.00ns INFO     cocotb                             Seeding Python random module with 1675619926
     0.00ns ERROR    cocotb.regression                  Configuring the assertion rewrite hook using pytest 7.2.0 failed. Please file a bug report!
                                                        Traceback (most recent call last):
                                                          File "/usr/ports/cad/py-cocotb/work-py39/stage/usr/local/lib/python3.9/site-packages/cocotb/regression.py", line 256, in _setup_pytest_assertion_rewriting
                                                            pytest_conf = Config.fromdictargs(
                                                          File "/usr/local/lib/python3.9/site-packages/_pytest/config/__init__.py", line 1110, in fromdictargs
                                                            config.parse(args, addopts=False)
                                                          File "/usr/local/lib/python3.9/site-packages/_pytest/config/__init__.py", line 1346, in parse
                                                            self._preparse(args, addopts=addopts)
                                                          File "/usr/local/lib/python3.9/site-packages/_pytest/config/__init__.py", line 1229, in _preparse
                                                            self.pluginmanager.load_setuptools_entrypoints("pytest11")
                                                          File "/usr/local/lib/python3.9/site-packages/pluggy/_manager.py", line 288, in load_setuptools_entrypoints
                                                            self.register(plugin, name=ep.name)
                                                          File "/usr/local/lib/python3.9/site-packages/_pytest/config/__init__.py", line 489, in register
                                                            ret: Optional[str] = super().register(plugin, name)
                                                          File "/usr/local/lib/python3.9/site-packages/pluggy/_manager.py", line 103, in register
                                                            hookimpl_opts = self.parse_hookimpl_opts(plugin, name)
                                                          File "/usr/local/lib/python3.9/site-packages/_pytest/config/__init__.py", line 460, in parse_hookimpl_opts
                                                            return _get_legacy_hook_marks(
                                                          File "/usr/local/lib/python3.9/site-packages/_pytest/config/__init__.py", line 374, in _get_legacy_hook_marks
                                                            warn_explicit_for(cast(FunctionType, method), message)
                                                          File "/usr/local/lib/python3.9/site-packages/_pytest/warning_types.py", line 171, in warn_explicit_for
                                                            raise type(w)(f"{w}\n at {filename}:{lineno}") from None
                                                        pytest.PytestDeprecationWarning: The hookimpl pytest_load_initial_conftests uses old-style configuration options (marks or attributes).
                                                        Please use the pytest.hookimpl(tryfirst=True) decorator instead
                                                         to configure the hooks.
                                                         See https://docs.pytest.org/en/latest/deprecations.html#configuring-hook-specs-impls-using-markers
                                                         at /usr/local/lib/python3.9/site-packages/pytest_cov/plugin.py:115
     0.00ns INFO     cocotb.regression                  Found test test_iteration_es.recursive_discovery
     0.00ns INFO     cocotb.regression                  Found test test_iteration_es.dual_iteration
     0.00ns INFO     cocotb.regression                  running recursive_discovery (1/2)
                                                          Recursively discover every single object in the design
```

Version: 1.7.2
Python-3.9
pytest-7.2.0
FreeBSD 13.1

marlonjames, 2023-02-05:
Looks like an issue with outdated `pytest-cov`. You are running `2.9.0` in #3230, so I am assuming this is the case here as well.

`pytest-cov` was updated to use the new-style hookimpl in `4.0.0`:
https://pytest-cov.readthedocs.io/en/latest/changelog.html#id1
Via Pull Request:
https://github.com/pytest-dev/pytest-cov/pull/549

Try with an updated `pytest-cov`.

yurivict, 2023-02-05:
With pytest-cov-4.0.0 tests don't fail any more, but Python errors are still present:
```
MODULE=issue_768 TESTCASE= TOPLEVEL=sample_module TOPLEVEL_LANG=verilog \
         /usr/local/bin/vvp -M /usr/ports/cad/py-cocotb/work-py39/stage/usr/local/lib/python3.9/site-packages/cocotb/libs -m libcocotbvpi_icarus   sim_build/sim.vvp 
     -.--ns INFO     gpi                                ..mbed/gpi_embed.cpp:78   in set_program_name_in_venv        Did not detect Python virtual environment. Using system-wide Python interpreter
     -.--ns INFO     gpi                                ../gpi/GpiCommon.cpp:101  in gpi_print_registered_impl       VPI registered
     0.00ns INFO     cocotb                             Running on Icarus Verilog version 11.0 (stable)
     0.00ns INFO     cocotb                             Running tests with cocotb v1.7.2 from /usr/ports/cad/py-cocotb/work-py39/stage/usr/local/lib/python3.9/site-packages/cocotb
     0.00ns INFO     cocotb                             Seeding Python random module with 1675634916
     0.00ns ERROR    cocotb.regression                  Configuring the assertion rewrite hook using pytest 7.2.0 failed. Please file a bug report!
                                                        Traceback (most recent call last):
                                                          File "/usr/ports/cad/py-cocotb/work-py39/stage/usr/local/lib/python3.9/site-packages/cocotb/regression.py", line 256, in _setup_pytest_assertion_rewriting
                                                            pytest_conf = Config.fromdictargs(
                                                          File "/usr/local/lib/python3.9/site-packages/_pytest/config/__init__.py", line 1110, in fromdictargs
                                                            config.parse(args, addopts=False)
                                                          File "/usr/local/lib/python3.9/site-packages/_pytest/config/__init__.py", line 1346, in parse
                                                            self._preparse(args, addopts=addopts)
                                                          File "/usr/local/lib/python3.9/site-packages/_pytest/config/__init__.py", line 1229, in _preparse
                                                            self.pluginmanager.load_setuptools_entrypoints("pytest11")
                                                          File "/usr/local/lib/python3.9/site-packages/pluggy/_manager.py", line 287, in load_setuptools_entrypoints
                                                            plugin = ep.load()
                                                          File "/usr/local/lib/python3.9/site-packages/setuptools/_vendor/importlib_metadata/__init__.py", line 194, in load
                                                            module = import_module(match.group('module'))
                                                          File "/usr/local/lib/python3.9/importlib/__init__.py", line 127, in import_module
                                                            return _bootstrap._gcd_import(name[level:], package, level)
                                                          File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
                                                          File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
                                                          File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
                                                          File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
                                                          File "/usr/local/lib/python3.9/site-packages/_pytest/assertion/rewrite.py", line 168, in exec_module
                                                            exec(co, module.__dict__)
                                                          File "/usr/local/lib/python3.9/site-packages/pytest_mock.py", line 162, in <module>
                                                            def mocker(pytestconfig):
                                                          File "/usr/local/lib/python3.9/site-packages/_pytest/fixtures.py", line 1338, in yield_fixture
                                                            warnings.warn(YIELD_FIXTURE, stacklevel=2)
                                                        pytest.PytestDeprecationWarning: @pytest.yield_fixture is deprecated.
                                                        Use @pytest.fixture instead; they are the same.
     0.00ns INFO     cocotb.regression                  Found test issue_768.do_test
     0.00ns INFO     cocotb.regression                  running do_test (1/1)
VCD info: dumpfile waveform.vcd opened for output.
sample_module: mybit has been updated, new value is 1
sample_module: mybits has been updated, new value is 11
     0.00ns INFO     cocotb.regression                  do_test passed
     0.00ns INFO     cocotb.regression                  **************************************************************************************
                                                        ** TEST                          STATUS  SIM TIME (ns)  REAL TIME (s)  RATIO (ns/s) **
                                                        **************************************************************************************
[... cut ...]
                                                        ** issue_892.error_test           PASS           1.00           0.00       1209.59  **
                                                        **************************************************************************************
                                                        ** TESTS=1 PASS=1 FAIL=0 SKIP=0                  1.00           0.45          2.24  **
                                                        **************************************************************************************
```

Version: 1.7.2
FreeBSD 13.1

marlonjames, 2023-02-06:
This error is due to outdated `pytest-mock`:

```
File "/usr/local/lib/python3.9/site-packages/pytest_mock.py", line 162, in <module>
  def mocker(pytestconfig):
File "/usr/local/lib/python3.9/site-packages/_pytest/fixtures.py", line 1338, in yield_fixture
  warnings.warn(YIELD_FIXTURE, stacklevel=2)
pytest.PytestDeprecationWarning: @pytest.yield_fixture is deprecated.
Use @pytest.fixture instead; they are the same.
```

I appreciate your effort in trying cocotb on FreeBSD, but please take the time to investigate the errors and ensure your testing environment doesn't have these incompatibilities.

FreeBSD is not officially supported (https://docs.cocotb.org/en/stable/platform_support.html), but I am interested in how cocotb fairs on the platform.

yurivict, 2023-02-06:
> FreeBSD is not officially supported (https://docs.cocotb.org/en/stable/platform_support.html), but I am interested in how cocotb fairs on the platform.

Thank you for your very positive attitude!

--------------------------------------------------------------

After the update to py39-pytest-mock-3.10.0  most Python errors disappeared, but some Python and other errors still remain, see the [log](https://people.freebsd.org/~yuri/py39-cocotb-1.7.2-tests.log).

marlonjames, 2023-02-06:
Some of the errors are expected and are handled in the Makefiles.

The failure of interest is this one.
It can't find `cocotb` module installed, and it can't find the `python` binary.
```
cd test_cases/test_custom_entry && gmake
gmake[4]: Entering directory '/usr/ports/cad/py-cocotb/work-py39/cocotb-1.7.2/tests/test_cases/test_custom_entry'
gmake all
gmake[5]: Entering directory '/usr/ports/cad/py-cocotb/work-py39/cocotb-1.7.2/tests/test_cases/test_custom_entry'
Traceback (most recent call last):
  File "/usr/ports/cad/py-cocotb/work-py39/stage/usr/local/bin/cocotb-config", line 5, in <module>
    from cocotb.config import main
ModuleNotFoundError: No module named 'cocotb'
../../designs/sample_module/Makefile:50: /Makefile.sim: No such file or directory
gmake[5]: *** No rule to make target '/Makefile.sim'.
gmake[5]: Failed to remake makefile '/Makefile.sim'.
gmake[5]: *** No rule to make target 'all'.
gmake[5]: Leaving directory '/usr/ports/cad/py-cocotb/work-py39/cocotb-1.7.2/tests/test_cases/test_custom_entry'
gmake[4]: [Makefile:6: override_for_this_test] Error 2 (ignored)
python -c 'import filecmp; assert filecmp.cmp("results.log", "expected_results.log")'
bash: line 1: python: command not found
gmake[4]: *** [Makefile:7: override_for_this_test] Error 127
gmake[4]: Leaving directory '/usr/ports/cad/py-cocotb/work-py39/cocotb-1.7.2/tests/test_cases/test_custom_entry'
gmake[3]: *** [Makefile:39: test_custom_entry] Error 2
```

yurivict, 2023-02-06:
> It can't find cocotb module installed, and it can't find the python binary.

The cocotb Python module is installed into a stage directory with PYTHONPATH pointing to it: ```PYTHONPATH=/usr/ports/cad/py-cocotb/work-py39/stage/usr/local/lib/python3.9/site-packages```

It looks like the testsuite loses the ```PYTHONPATH``` variable and only Python's default module locations are left, where ```cocotb``` isn't installed.

> it can't find the python binary

The ```python``` executable was indeed missing. I added such link to the PATH.

Here is the updated [log](https://people.freebsd.org/~yuri/py39-cocotb-1.7.2-tests-1.log).

marlonjames, 2023-02-08:
So I think due to your approach with `PYTHONPATH`, that test will always fail.

closed 2023-03-18 by github-actions[bot]