cocotb/cocotb #3245: Array inputs not working in xcelium/incisive

davekeeshan, 2023-02-14:
I found this in a bigger design, but I have reduced it down to this. This may or may not be related to these issues, #3239 and #2982

When I have a array input, `input wire [7:0] rgb [2:0]`, cocotb can't seem to plug into either of the cadence tools, xcelium or incisive (ius).

I have written and checked in this example, [test_array_xcelium](https://github.com/daxzio/test_array_xcelium).

This test passes when run under icarus.

It fails on xcelium and incisive, as their native tools, (there is an incisive mode in xcelium, I am not checking that)

This is what I expect (and get as an out in icarus):

```
pixel2 = 10011000
pixel1 = 10111010
pixel0 = 11011100
```

This is what I get from xecelium:

```
pixel2 = xxxxxxxx
pixel1 = xxxxxxxx
pixel0 = 01100000
    33.00ns DEBUG    gpi                                Not removing PRIMED timer 5500
    33.00ns INFO     cocotb.regression                  test_array failed
                                                        Traceback (most recent call last):
                                                          File "/home/xuser/projects/test_array_xcelium/tb_dut.py", line 69, in test_array
                                                            assert tb.cr.pixel2.value == 0x98
                                                          File "/home/xuser/.venv/Python-v3.10.7/lib/python3.10/site-packages/cocotb/binary.py", line 544, in __eq__
                                                            return self.value == other
                                                          File "/home/xuser/.venv/Python-v3.10.7/lib/python3.10/site-packages/cocotb/binary.py", line 381, in value
                                                            return self.integer
                                                          File "/home/xuser/.venv/Python-v3.10.7/lib/python3.10/site-packages/cocotb/binary.py", line 372, in integer
                                                            return self._convert_from(self._str)
                                                          File "/home/xuser/.venv/Python-v3.10.7/lib/python3.10/site-packages/cocotb/binary.py", line 252, in _convert_from_unsigned
                                                            return int(x.translate(_resolve_table), 2)
                                                          File "/home/xuser/.venv/Python-v3.10.7/lib/python3.10/site-packages/cocotb/binary.py", line 84, in __missing__
                                                            return self.resolve_x(key)
                                                          File "/home/xuser/.venv/Python-v3.10.7/lib/python3.10/site-packages/cocotb/binary.py", line 61, in resolve_error
                                                            raise ValueError(
                                                        ValueError: Unresolvable bit in binary string: 'x'
```

the x's on 2 and 1 are obviously wrong, but even the value coming out for pixel0 is also incorrect.

Having a look in waves we see that the values that got placed on RGB are wrong (so that the returned pixels values are the _right_ wrong, they would correctly be these outputs if the inputs were explicitly set to what rgb is being set to)

This is what waves look like, `rgb[0]` should be `0x23` not `0x9f`

[image]

I was told to look at this test, [test_array](https://github.com/cocotb/cocotb/tree/master/tests/test_cases/test_array), in the cocotb test suite.  When I run it locally, `make SIM=xcelium`, it fails for me:

```
     0.00ns INFO     cocotb.regression                  running test_read_write (1/6)
                                                          Test handle inheritance
    10.00ns INFO     cocotb.test                        Checking Generics/Parameters:
    10.00ns INFO     cocotb.test                           Found ConstantObject(array_module.param_logic) (GPI_INTEGER) with value=0x1
    10.00ns INFO     cocotb.test                           Found ConstantObject(array_module.param_logic_vec) (GPI_INTEGER) with value=0xDA
    10.00ns INFO     cocotb.test                        Checking Constants:
    10.00ns INFO     cocotb.test                           Found ConstantObject(array_module.const_logic) (GPI_INTEGER) with value=0x0
    10.00ns INFO     cocotb.test                           Found ConstantObject(array_module.const_logic_vec) (GPI_INTEGER) with value=0x3D
    20.00ns INFO     cocotb.test                        Writing the signals!!!
    30.00ns INFO     cocotb.test                        Checking writes:
    30.00ns INFO     cocotb.test                           Found ModifiableObject(array_module.port_logic_out) (GPI_REGISTER) with value=0x1
    30.00ns INFO     cocotb.test                           Found ModifiableObject(array_module.port_logic_vec_out) (GPI_REGISTER) with value=0xCC
    30.00ns INFO     cocotb.regression                  test_read_write failed
                                                        Traceback (most recent call last):
                                                          File "/home/xuser/projects/cocotb/tests/test_cases/test_array/test_array.py", line 162, in test_read_write
                                                            _check_value(tlog, dut.sig_t2, [0xCC, 0xDD, 0xEE, 0xFF])
                                                          File "/home/xuser/projects/cocotb/tests/test_cases/test_array/test_array.py", line 56, in _check_value
                                                            assert (
                                                          File "/home/xuser/projects/cocotb/cocotb/binary.py", line 544, in __eq__
                                                            return self.value == other
                                                          File "/home/xuser/projects/cocotb/cocotb/binary.py", line 381, in value
                                                            return self.integer
                                                          File "/home/xuser/projects/cocotb/cocotb/binary.py", line 372, in integer
                                                            return self._convert_from(self._str)
                                                          File "/home/xuser/projects/cocotb/cocotb/binary.py", line 252, in _convert_from_unsigned
                                                            return int(x.translate(_resolve_table), 2)
                                                          File "/home/xuser/projects/cocotb/cocotb/binary.py", line 84, in __missing__
                                                            return self.resolve_x(key)
                                                          File "/home/xuser/projects/cocotb/cocotb/binary.py", line 61, in resolve_error
                                                            raise ValueError(
                                                        ValueError: Unresolvable bit in binary string: 'z'

```
cocotb version: 1.7.2
operating system: redhat8_x64
tool: XCELIUM2209/004
python: system 3.10.7

spetca, 2023-02-15:
I only have icarus on my machine, but does the output of `make COCOTB_LOG_LEVEL=DEBUG` with xcelium differ from this? interested in the report on the array sizes 

output from icarus: 
```
     0.00ns DEBUG    gpi                                Searching for clk
     0.00ns DEBUG    gpi                                VPI: clk initialized with 1 elements
     0.00ns DEBUG    gpi                                VPI: Created GPI object from type vpiNet(36)
     0.00ns DEBUG    gpi                                Checking dut.clk exists
     0.00ns DEBUG    cocotb.clk                         Created
     0.00ns DEBUG    gpi                                Searching for rgb
     0.00ns DEBUG    gpi                                VPI: Created GPI object from type vpiNetArray(114)
     0.00ns DEBUG    gpi                                Checking dut.rgb exists
     0.00ns DEBUG    cocotb.rgb                         Created
     0.00ns DEBUG    gpi                                Searching for pixel0
     0.00ns DEBUG    gpi                                VPI: Indexable object initialized with range [7:0] and length >8<
     0.00ns DEBUG    gpi                                VPI: pixel0 initialized with 8 elements
     0.00ns DEBUG    gpi                                VPI: Created GPI object from type vpiNet(36)
     0.00ns DEBUG    gpi                                Checking dut.pixel0 exists
     0.00ns DEBUG    cocotb.pixel0                      Created
     0.00ns DEBUG    gpi                                Searching for pixel1
     0.00ns DEBUG    gpi                                VPI: Indexable object initialized with range [7:0] and length >8<
     0.00ns DEBUG    gpi                                VPI: pixel1 initialized with 8 elements
     0.00ns DEBUG    gpi                                VPI: Created GPI object from type vpiNet(36)
     0.00ns DEBUG    gpi                                Checking dut.pixel1 exists
     0.00ns DEBUG    cocotb.pixel1                      Created
     0.00ns DEBUG    gpi                                Searching for pixel2
     0.00ns DEBUG    gpi                                VPI: Indexable object initialized with range [7:0] and length >8<
     0.00ns DEBUG    gpi                                VPI: pixel2 initialized with 8 elements
     0.00ns DEBUG    gpi                                VPI: Created GPI object from type vpiNet(36)
     0.00ns DEBUG    gpi                                Checking dut.pixel2 exists
     0.00ns DEBUG    cocotb.pixel2                      Created
     0.00ns DEBUG    gpi                                clk has 1 elements
VCD info: dumpfile dut.vcd opened for output.
    11.00ns DEBUG    gpi                                Checking if index 0 native through implementation VPI 
    11.00ns DEBUG    gpi                                VPI: Indexable object initialized with range [7:0] and length >8<
    11.00ns DEBUG    gpi                                VPI: rgb[0] initialized with 8 elements
    11.00ns DEBUG    gpi                                VPI: Created GPI object from type vpiNet(36)
    11.00ns DEBUG    gpi                                Checking dut.rgb[0] exists
    11.00ns DEBUG    cocotb.rgb[0]                      Created
    11.00ns DEBUG    gpi                                rgb[0] has 8 elements
    11.00ns DEBUG    gpi                                Checking if index 1 native through implementation VPI 
    11.00ns DEBUG    gpi                                VPI: Indexable object initialized with range [7:0] and length >8<
    11.00ns DEBUG    gpi                                VPI: rgb[1] initialized with 8 elements
    11.00ns DEBUG    gpi                                VPI: Created GPI object from type vpiNet(36)
    11.00ns DEBUG    gpi                                Checking dut.rgb[1] exists
    11.00ns DEBUG    cocotb.rgb[1]                      Created
    11.00ns DEBUG    gpi                                rgb[1] has 8 elements
    11.00ns DEBUG    gpi                                Checking if index 2 native through implementation VPI 
    11.00ns DEBUG    gpi                                VPI: Indexable object initialized with range [7:0] and length >8<
    11.00ns DEBUG    gpi                                VPI: rgb[2] initialized with 8 elements
    11.00ns DEBUG    gpi                                VPI: Created GPI object from type vpiNet(36)
    11.00ns DEBUG    gpi                                Checking dut.rgb[2] exists
    11.00ns DEBUG    cocotb.rgb[2]                      Created
    11.00ns DEBUG    gpi                                rgb[2] has 8 elements
```

davekeeshan, 2023-02-15:
@spetca 

I forgot I had actually looked at this, also icarus versus xcelium, but to my eye they looked the same:

```
     0.00ns INFO     cocotb.regression                  running test_array (1/1)
     0.00ns DEBUG    gpi                                Searching for clk
     0.00ns DEBUG    gpi                                VPI: clk initialized with 1 elements
     0.00ns DEBUG    gpi                                VPI: Created GPI object from type vpiNet(36)
     0.00ns DEBUG    gpi                                Checking dut.clk exists
     0.00ns DEBUG    cocotb.clk                         Created
     0.00ns DEBUG    gpi                                Searching for rgb
     0.00ns DEBUG    gpi                                VPI: Created GPI object from type vpiNetArray(114)
     0.00ns DEBUG    gpi                                Checking dut.rgb exists
     0.00ns DEBUG    cocotb.rgb                         Created
     0.00ns DEBUG    gpi                                Searching for pixel0
     0.00ns DEBUG    gpi                                VPI: Indexable object initialized with range [7:0] and length >8<
     0.00ns DEBUG    gpi                                VPI: pixel0 initialized with 8 elements
     0.00ns DEBUG    gpi                                VPI: Created GPI object from type vpiNet(36)
     0.00ns DEBUG    gpi                                Checking dut.pixel0 exists
     0.00ns DEBUG    cocotb.pixel0                      Created
     0.00ns DEBUG    gpi                                Searching for pixel1
     0.00ns DEBUG    gpi                                VPI: Indexable object initialized with range [7:0] and length >8<
     0.00ns DEBUG    gpi                                VPI: pixel1 initialized with 8 elements
     0.00ns DEBUG    gpi                                VPI: Created GPI object from type vpiNet(36)
     0.00ns DEBUG    gpi                                Checking dut.pixel1 exists
     0.00ns DEBUG    cocotb.pixel1                      Created
     0.00ns DEBUG    gpi                                Searching for pixel2
     0.00ns DEBUG    gpi                                VPI: Indexable object initialized with range [7:0] and length >8<
     0.00ns DEBUG    gpi                                VPI: pixel2 initialized with 8 elements
     0.00ns DEBUG    gpi                                VPI: Created GPI object from type vpiNet(36)
     0.00ns DEBUG    gpi                                Checking dut.pixel2 exists
     0.00ns DEBUG    cocotb.pixel2                      Created
     0.00ns DEBUG    gpi                                clk has 1 elements
    11.00ns DEBUG    gpi                                Checking if index 0 native through implementation VPI 
    11.00ns DEBUG    gpi                                VPI: Indexable object initialized with range [7:0] and length >8<
    11.00ns DEBUG    gpi                                VPI: rgb[0] initialized with 8 elements
    11.00ns DEBUG    gpi                                VPI: Created GPI object from type vpiNet(36)
    11.00ns DEBUG    gpi                                Checking dut.rgb[0] exists
    11.00ns DEBUG    cocotb.rgb[0]                      Created
    11.00ns DEBUG    gpi                                rgb[0] has 8 elements
    11.00ns DEBUG    gpi                                Checking if index 1 native through implementation VPI 
    11.00ns DEBUG    gpi                                VPI: Indexable object initialized with range [7:0] and length >8<
    11.00ns DEBUG    gpi                                VPI: rgb[1] initialized with 8 elements
    11.00ns DEBUG    gpi                                VPI: Created GPI object from type vpiNet(36)
    11.00ns DEBUG    gpi                                Checking dut.rgb[1] exists
    11.00ns DEBUG    cocotb.rgb[1]                      Created
    11.00ns DEBUG    gpi                                rgb[1] has 8 elements
    11.00ns DEBUG    gpi                                Checking if index 2 native through implementation VPI 
    11.00ns DEBUG    gpi                                VPI: Indexable object initialized with range [7:0] and length >8<
    11.00ns DEBUG    gpi                                VPI: rgb[2] initialized with 8 elements
    11.00ns DEBUG    gpi                                VPI: Created GPI object from type vpiNet(36)
    11.00ns DEBUG    gpi                                Checking dut.rgb[2] exists
    11.00ns DEBUG    cocotb.rgb[2]                      Created
    11.00ns DEBUG    gpi                                rgb[2] has 8 elements
```

ktbarrett, 2024-08-17:
`test_cocotb_array` tests indexing into the `in_2d_vect_packed_unpacked` defined in the same way as `rgb` in your test, and does not have a problem in our CI. Granted our CI is testing against a newer version of Xcelium: 23.09. Try updating.

```verilog
  input     logic               [2:0]               in_2d_vect_packed_unpacked[2:0]                     ,
```

closed 2024-09-24 by github-actions[bot]