cocotb/cocotb #3249: TWOS_COMPLIMENT binary representation is not giving correct answer

RehanQasim-dev, 2023-02-16:
I have just learning cocotb and I am trying to get BinaryValue representation of int 6 in 2's compliment but i am not getting correct answer it should be 1010 but i am getting 0110,

async def simple_test(dut):
    """simple test of 1+2"""
    A=BinaryValue(6,binaryRepresentation=BinaryRepresentation.TWOS_COMPLEMENT)
    print("TWO's compliment of 6 is=",A)

[image]

eric-wieser, 2023-02-16:
Why would you expect `1010`? That looks like `-6` or `10` to me.

RehanQasim-dev, 2023-02-16:
> Why would you expect `1010`? That looks like `-6` or `10` to me.
00110 is the binary representation of 6. To find its two's complement I take its negation and add 1 which comes out to be 11010

eric-wieser, 2023-02-16:
"Twos complement" here descrives the representation, not an operation. If you want to get it's negation, use `-`.

closed 2023-02-18 by ktbarrett