cocotb/cocotb #3254: Verilator: Support `--timing` mode

marlonjames, 2023-02-21:
Verilator has support for [timing constructs](https://verilator.org/guide/latest/exe_verilator.html#cmdoption-timing) via the `--timing` argument.

When using `--timing`, the wrapper needs to call `eventsPending()` and `nextTimeSlot()` in order to not miss a time step that needs to be evaluated.
Details here: https://verilator.org/guide/latest/connecting.html#wrappers-and-model-evaluation-loop

imphil, 2023-02-22:
I wonder if we can use the auto-generated C++ wrapper with Verilator now, which would hopefully give us all these things for free (and keep the updated).

marlonjames, 2023-02-23:
I don't think that the generated wrapper has support for any of the callbacks for VPI applications.