cocotb/cocotb #3257: `RisingEdge` in Questa and VHDL `rising_edge` have different semantics

ktbarrett, 2023-02-23:
`RisingEdge` is essentially equivalent in VHDL to `some_signal'event and To_X01(some_signal) = '1'`, which has different semantics than the builtin VHDL function `rising_edge` which has the semantics `some_signal'event and To_X01(some_signal) = '1' and To_X01(some_signal'last_value) = '0'`. This causes difference of operation usually at the start of the sim when signals transition from `'U'` to `'1'`, or from `'U'` to `'0'` in the `FallingEdge` case.

ktbarrett, 2023-02-23:
Quick workaround.

```python
async def true_rising_edge(clk) -> None:
    while True:
        last_0 = clk.value.binstr == "0"
        await RisingEdge(clk)
        if last_0:
            break
```

marlonjames, 2023-02-23:
`'last_value` is available through VHPI, if you wanted to match the semantics.