cocotb/cocotb #3259: Verilator becomes unresponsive to CocoTB after instantiating cocotb_bus::Bus class

apstrike, 2023-03-02:
This is something I found with:
- OS Fedora 29 (64-bit)
- python V3.7.2
-  cocotb V1.7.2
- Verilator 5.006 (build with gcc version 8.2.1 20181215 (Red Hat 8.2.1-6) (GCC))

Verilator seems to be unresponsive (signal's don't change according to setting the signal values) to CocoTB after instantiating a `cocotb_bus::Bus class`. I've been able to work around this by setting the `case_insensitivty` flag to False. 

A pretty simple example of this can be seen below:
Makefile:
```
# locate the cocotb installation
COCOTB:=$(shell cocotb-config --share)

export COCOTB_LOG_LEVEL=DEBUG
export PYTHON_BIN=python3

export SIM=verilator
EXTRA_ARGS += -Wno-fatal
EXTRA_ARGS += --trace

VERILOG_SOURCES :=  mymodule.sv

TOPLEVEL:=mymodule

TOPLEVEL_LANG:=verilog
MODULE:=mymodule_cocotb

include $(COCOTB)/makefiles/Makefile.inc
include $(COCOTB)/makefiles/Makefile.sim
```

SV Module file:
```
module mymodule (
    input  wire         clk_i,
    input  wire         reset_i,

    input  wire         mybus_read,
    input  wire         mybus_write,
    input  wire  [31:0] mybus_address,
    input  wire  [31:0] mybus_writedata,
    output logic        mybus_readdatavalid,
    output logic [31:0] mybus_readdata
);

    always_ff @(posedge clk_i) begin
        if (reset_i) begin
            mybus_readdatavalid <= 1'b0;
        end else begin
            mybus_readdatavalid <= 1'b0;
            mybus_readdata <= 32'b0;
            if (mybus_read) begin
                mybus_readdatavalid <= 1'b1;
                mybus_readdata <= 32'hDEADBEEF;
            end
        end
    end
endmodule
```

cocotb test file:
```
import cocotb
from cocotb.triggers import Timer, RisingEdge, ReadOnly, ClockCycles
from cocotb.clock import Clock
from cocotb_bus.bus import Bus

_signals = ["address"]
_optional_signals = ["readdata", "read", "write",
[... 270 lines cut ...]
     0.00ns DEBUG    cocotb.mymodule                    Signal name mybus_readdatavalid
     0.00ns DEBUG    cocotb.mymodule                    Signal name mybus_readdatavalid, idx None
     0.00ns DEBUG    gpi                                reset_i has 1 elements
     0.00ns DEBUG    gpi                                clk_i has 1 elements
    29.45ns DEBUG    gpi                                Not removing PRIMED timer 1550
    29.45ns INFO     cocotb.regression                  failing_test failed
                                                        Traceback (most recent call last):
                                                          File "/home/myuser/tmp/cocotb_verilator/mymodule_cocotb.py", line 29, in failing_test
                                                            assert dut.reset_i.value.integer == 1
                                                        AssertionError
    29.45ns INFO     cocotb.regression                  **************************************************************************************
                                                        ** TEST                          STATUS  SIM TIME (ns)  REAL TIME (s)  RATIO (ns/s) **
                                                        **************************************************************************************
                                                        ** mymodule_cocotb.failing_test   FAIL          29.45           0.00       7109.85  **
                                                        **************************************************************************************
                                                        ** TESTS=1 PASS=0 FAIL=1 SKIP=0                 29.45           0.01       2607.64  **
                                                        **************************************************************************************
                                                        
- :0: Verilog $finish
make[1]: Leaving directory '/home/myuser/tmp/cocotb_verilator'
```

To be honest, I'm not sure if this is an issue with Verilator or Cocotb, I haven't been able to understand much in regards to how Verialtor and CocoTB interact.

I have determined that avoiding calling `cocotb_bus/bus.pt::Bus:def _caseInsensGetattr` works around the issue.

marlonjames, 2023-03-02:
You are not the first to hit this. The problem is tracked in #2244

apstrike, 2023-03-03:
I see, thanks.

closed 2023-03-03 by apstrike