cocotb/cocotb #3267: RecursionError in _resume_coro_upon when error happen

suzizecat, 2023-03-17:
Using :
 - Cocotb 1.7.2
 - Ubuntu 22.04 (In WSL 2.0 for Windows 10) 
 - Verilator 5.004

When an exception occurs in the scheduler `_react` function on:
```python
        if self._pending_triggers:
            raise InternalError(
                "Expected all triggers to be handled but found {}".format(
                    self._pending_triggers
                )
            )
```
With remaining triggers, during the call to the `trigger.prime` function of the `_resume_coro_upon()` :
```python
if not trigger.primed:
    # ... 
    try:
        trigger.prime(self._react)
    except Exception as e:
        # discard the trigger we associated, it will never fire
        self._trigger2coros.pop(trigger)
        # replace it with a new trigger that throws back the exception
        self._resume_coro_upon(
            coro,
            NullTrigger(
                name="Trigger.prime() Error", outcome=outcomes.Error(e)
            ),
        )
```

The call of `self._resume_coro_upon` with the `NullTrigger` will end up throwing the same `InternalError`, hence recursively throwing until Python itself explodes.

I managed to avoid the recursion error by changing the code around `InternalError`. Adding a `self._cleanup()` force the scheduler to cleanup the remaining triggers and fail in another way, after properly reporting the `InternalError`. 

The resulting code for `_react` is:
```python
def _react(self, trigger):
    """
    Called when a trigger fires.
    We ensure that we only start the event loop once, rather than
    letting it recurse.
    """
    # ... 
    if self._pending_triggers:
        self._cleanup() # <--- Force the cleanup of remaining _pending_triggers
        raise InternalError(
            "Expected all triggers to be handled but found {}".format(
                self._pending_triggers
            )
        )
```

ktbarrett, 2023-03-22:
I think instead `InternalError` should either specifically handled in the wrapper around `trigger.prime` or it should become a `BaseException`.

ktbarrett, 2023-04-06:
@suzizecat Can you provide a test for this?

ktbarrett, 2023-04-06:
I used the test from #3270 to reproduce the issue. Changing to BaseException worked, I'll put up a PR once I figure out the rest of #3270.

suzizecat, 2023-04-10:
The test from my other issue is indeed the most reliable (and simple-ish) way I found to trigger this.

PR #3291 (2023-05-12): Fix scheduler bug
Fixes #3267 and #3270. The shallow copy of `_triggers2coros` and the cleanup of triggers (in the `_event_loop` loop) after coros (in `_cleanup`) cause #3270. #3267 is dealt with by making `InternalError` a `BaseException` as was discussed in that issue.
closed 2023-05-12 by ktbarrett