cocotb/cocotb #3268: Support custom types in VHDL

sigmuha, 2023-03-19:
Add support for custom types in VHDL.
It would be nice to access types (enumerated and physical) either via the handle of a signal of that type or otherwise. 
As of now, I don't have any other way than doing my own conversion with a dict in python.

Versions:
- cocotb: 1.7.2
- simulator: GHDL 4.0.0-dev (3.0.0.r15.gc0975bc8a.dirty) [Dunoon edition]
- python version: 3.8.6

I found this [duplicate issue](https://github.com/cocotb/cocotb/issues/2014) which requested the same feature but is from an older version. There they stated it was possible to access it via the handle of a signal. The syntax was described as:
```python
handle = dut['phys_signal']
distance = handle.get_type()
```
This syntax isn't possible in the current version of cocotb. A similar syntax, but with updated access to the handle, would be great! As far as I understand, it is accessible from the vhpi?

closed 2023-03-22 by marlonjames

cmarqu, 2023-03-24:
@sigmuha Maybe there is a misunderstanding about the old issue - it is not claimed there that `handle.get_type()` is a working feature, but a possible way that a user interface *could* look, if someone would be implementing it.

sigmuha, 2023-03-25:
@cmarqu ah, I see. 
Could this be tagged as a feature request instead?