cocotb/cocotb #3284: Hello World example from README does not terminate in verilator

hansfbaier, 2023-04-21:
OS: 64 bit
Distributor ID:	Ubuntu
Description:	Ubuntu 22.04.2 LTS

python: python3 stock Ubuntu
Python 3.10.6 (main, Mar 10 2023, 10:55:28) [GCC 11.3.0] on linux
>>> 

```
$ make SIM=verilator
[...]
MODULE=test_dff TESTCASE= TOPLEVEL=dff TOPLEVEL_LANG=verilog \
         sim_build/Vtop 
     -.--ns INFO     gpi                                ..mbed/gpi_embed.cpp:76   in set_program_name_in_venv        Did not detect Python virtual environment. Using system-wide Python interpreter
     -.--ns INFO     gpi                                ../gpi/GpiCommon.cpp:101  in gpi_print_registered_impl       VPI registered
     0.00ns INFO     cocotb                             Running on Verilator version 5.004 2022-12-14
     0.00ns INFO     cocotb                             Running tests with cocotb v1.7.2 from /home/jack/.local/lib/python3.10/site-packages/cocotb
     0.00ns INFO     cocotb                             Seeding Python random module with 1682072626
     0.00ns INFO     cocotb.regression                  Found test test_dff.test_dff_simple
     0.00ns INFO     cocotb.regression                  running test_dff_simple (1/1)
                                                          Test that d propagates to q
```

marlonjames, 2023-04-21:
> `Running on Verilator version 5.004 2022-12-14`

cocotb supports 5.006 or later. Try using a newer version.
https://docs.cocotb.org/en/latest/simulator_support.html#verilator

hansfbaier, 2023-04-21:
@marlonjames Oh thanks, that fixed it!

hansfbaier, 2023-04-21:
Would be nice to put this into a python assert.

closed 2023-04-21 by hansfbaier