cocotb/cocotb #3286: The `test_dff.py` example does not fail for a non-working DFF

ddribin, 2023-04-26:
The `test_dff.py` example in `examples/simple_dff/` does not fail for a non-working DFF. Changing `dff.sv`
to just pass through `d` without registering it still passes the test:

```verilog
module dff (
  input logic clk, d,
  output logic q
);

assign q = d;

endmodule
```

As this is obviously not a flip fop, it should fail, but it still passes:

```
     0.00ns INFO     cocotb.regression                  Found test test_dff.dff_simple_test
     0.00ns INFO     cocotb.regression                  running dff_simple_test (1/1)
                                                          Test that d propagates to q
105000.00ns INFO     cocotb.regression                  dff_simple_test passed
105000.00ns INFO     cocotb.regression                  **************************************************************************************
                                                        ** TEST                          STATUS  SIM TIME (ns)  REAL TIME (s)  RATIO (ns/s) **
                                                        **************************************************************************************
                                                        ** test_dff.dff_simple_test       PASS      105000.00           0.00   49886942.02  **
                                                        **************************************************************************************
                                                        ** TESTS=1 PASS=1 FAIL=0 SKIP=0             105000.00           0.23     455049.98  **
                                                        **************************************************************************************
```

- cocotb version: 1.7.2
- OS: macOS Monterey (12.6.5) (Intel 64-bit)
- Simulator: Icarus Verilog version 11.0
- Python version: 3.11.3, installed via [Homebrew](https://brew.sh)

ktbarrett, 2023-04-30:
The checker should be more strict. On a rising edge it should assert the new value is on the input, and the previous value is output. Then after waiting for the end of the time cycle (using `ReadOnly`) it should check the output has updated to the new value. It should also check to ensure the value does not change until after the next rising edge of the clock. I think that is a sufficient description of the functionality of a DFF, right?

ddribin, 2023-05-02:
Yeah, that seems accurate. That said, this is likely the first example folks new to `cocotb` are going to see. Folks are going to copy and paste this into their own tests, so it should really highlight best practices. So rather than trying to verify aspects like "the output value does not change until after the next rising edge", I may suggest boiling this down to a more simple description: On the rising edge, the previous clock's input value is on the output. That's the gist of sequential logic.

Also, I believe using `ReadOnly` has been problematic, and is likely not something that should be encouraged? I found some discussion on `ReadOnly` in `cocotb-bus` [Issue #26](https://github.com/cocotb/cocotb-bus/issues/26).

What do you think about this as a test?

```python
@cocotb.test()
async def dff_simple_test(dut):
    clock = Clock(dut.clk, 10, units="us")  # Create a 10us period clock on port clk
    cocotb.start_soon(clock.start())  # Start the clock

    await RisingEdge(dut.clk)  # Synchronize with the clock
    expected_val = None # Logic('z') does not work?
    for i in range(10):
        val = random.randint(0, 1)
        dut.d.value = val  # Assign the random value val to the input port d
        await RisingEdge(dut.clk)
        if expected_val != None:
            assert dut.q.value == expected_val, f"output q was incorrect on the {i}th cycle"
        expected_val = val # Save the value for the next clock

    # Check the final input on the next clock
    await RisingEdge(dut.clk)
    assert dut.q.value == expected_val, f"output q was incorrect on the last cycle"
```

I couldn't figure out how to assert the initial `z` value, so I just skipped it using `None`.

ddribin, 2023-05-03:
> Initial values are 'X' not 'Z'

That's what I would have thought, too, but it fails:

```
 10000.00ns INFO     cocotb.regression                  dff_simple_test failed
                                                        Traceback (most recent call last):
                                                          File "/Users/dave/work/cocotb/examples/simple_dff/test_dff.py", line 28, in dff_simple_test
                                                            assert LogicArray(dut.q.value) == expected_val, f"output q was incorrect on the {i}th cycle"
                                                        AssertionError: output q was incorrect on the 0th cycle
                                                        assert LogicArray('Z', Range(0, 'downto', 0)) == LogicArray('X', Range(0, 'downto', 0))
                                                         +  where LogicArray('Z', Range(0, 'downto', 0)) = LogicArray(z)
                                                         +    where z = ModifiableObject(dff.q).value
                                                         +      where ModifiableObject(dff.q) = HierarchyObject(dff).q
```

However `expected_val = LogicArray("Z")` passes. I'm not sure I understand what's going wrong here, but `GTKWave` output corroborates this:

[image]

I also tried setting an initial value in `dff.sv`, to avoid the issue:

```verilog
module dff (
  input logic clk, d,
  output logic q
);

initial q = 0;
always @(posedge clk) begin
  q <= d;
end
```

Then we could avoid `LogicArray` altogether by using:

```python
@cocotb.test()
async def dff_simple_test_with_initial(dut):
    """Test that d propagates to q"""

    clock = Clock(dut.clk, 10, units="us")  # Create a 10us period clock on port clk
    cocotb.start_soon(clock.start())  # Start the clock

    await RisingEdge(dut.clk)  # Synchronize with the clock
    expected_val = 0 # Initial value
    for i in range(10):
        val = random.randint(0, 1)
        dut.d.value = val  # Assign the random value val to the input port d
        await RisingEdge(dut.clk)
        assert dut.q.value == expected_val, f"output q was incorrect on the {i}th cycle"
        expected_val = val

    # Check the final input on the next clock
    await RisingEdge(dut.clk)
    assert dut.q.value == expected_val, f"output q was incorrect on the last cycle"
```

But this *still* expects a `Z` initial output. At this point I am really confused. Is this a quirk of Icarus Verilog? I know I *have* seen `X` outputs from Icarus before.

```
 10000.00ns INFO     cocotb.regression                  dff_simple_test_with_initial failed
                                                        Traceback (most recent call last):
                                                          File "/Users/dave/work/cocotb/examples/simple_dff/test_dff.py", line 47, in dff_simple_test_with_initial
                                                            assert dut.q.value == expected_val, f"output q was incorrect on the {i}th cycle"
                                                          File "/Users/dave/work/cocotb/cocotb/binary.py", line 544, in __eq__
                                                            return self.value == other
                                                                   ^^^^^^^^^^
                                                          File "/Users/dave/work/cocotb/cocotb/binary.py", line 381, in value
                                                            return self.integer
                                                                   ^^^^^^^^^^^^
                                                          File "/Users/dave/work/cocotb/cocotb/binary.py", line 372, in integer
                                                            return self._convert_from(self._str)
                                                                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
                                                          File "/Users/dave/work/cocotb/cocotb/binary.py", line 252, in _convert_from_unsigned
                                                            return int(x.translate(_resolve_table), 2)
                                                                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^
                                                          File "/Users/dave/work/cocotb/cocotb/binary.py", line 84, in __missing__
                                                            return self.resolve_x(key)
                                                                   ^^^^^^^^^^^^^^^^^^^
                                                          File "/Users/dave/work/cocotb/cocotb/binary.py", line 61, in resolve_error
                                                            raise ValueError(
                                                        ValueError: Unresolvable bit in binary string: 'z'
```

I've pushed up a branch in my fork, [`simple_dff_wip`][simple_dff_wip], if you want to try it out yourself.

[simple_dff_wip]: https://github.com/ddribin/cocotb/tree/simple_dff_wip/examples/simple_dff

ktbarrett, 2023-05-04:
I'll try this in some other simulators and get back to you.

ddribin, 2023-05-04:
I understand what is going on, now. You can see it more clearly by starting the clock low with `clock.start(start_high=False)`:

[image]

This shows the initial `q` value *is* `X`, but the initial value for `d` is apparently `Z`. So on that first `RisingClock` for `# Synchronize with the clock`, it registers that `Z`.

Not sure if the `d` input starting at `Z` is Icarus, specific, but I'll rework the test to take this into account, by either setting `d` before the clock starts and/or skipping that initial `RisingClock`.

ddribin, 2023-05-05:
Opened pull request https://github.com/cocotb/cocotb/pull/3294. If that looks good, I can update the example in `README.md`, too.

commit eab8ea90 (2023-05-11): Update simple_dff example to assert values on the rising edge. Fixes #3286 (#3294)
closed 2023-05-11 by ktbarrett