cocotb/cocotb #3292: An object is not found initially, but found if I print(dir(its parent object)), questa sim

akadours22, 2023-05-02:
Versions:
* cocotb 1.7.1 or 1.7.2 or repo master at 337b0d, all installed with pip3
* OS ubuntu 18 or 22 both 64 bit 
* Questa Intel Starter FPGA Edition-64 2021.2 Revision: 2021.04 Date: Apr 14 2021 with a free license
* Python 3.10.6 from apt-get 

How to reproduce:
Save this verilog to file `b.v`
```
module a
   (input wire clk);
   genvar i;
   generate
      for (i = 0; i < 1; i = i + 1) begin : d
         wire d_valid;
      end
   endgenerate
endmodule

module b
  (input wire clk);
   genvar i;
   generate
     for (i = 0; i < 1; i = i + 1) begin : c
       reg dummy;
	 a inst(clk);
       end
    endgenerate
 endmodule
```

Save this test to `b.py`
```
import cocotb
from cocotb.handle import Force

@cocotb.test()
async def atest(dut):
#    print(dir(dut.c[0].inst))
    dut.c[0].dummy = Force(0)
    dut.c[0].inst.d[0].d_valid = Force(0)
```

`vlib work`, `vlog b.v`, `vsim -do "run -all" -c  -pli `cocotb-config --lib-name-path vpi questa`  b`
Should give you "AttributeError: inst contains no object named d"

But if you uncomment the print in the `b.py`, it should pass. Always works good with icarus.

Looks like a bug.

akadours22, 2023-05-02:
but using `dut.c[0].inst.__getattr__("d[0]").d_valid.value = Force(0)` works

akadours22, 2023-05-03:
Ok, maybe it looks like a bug in questa, basically, it can find `b.c`, but it cannot find `b.c[0].inst.d` in my example with this VPI .cpp code:
```
#include <iostream>
#include <vpi_user.h>

using namespace std;

void printErrorIfAny() {
  t_vpi_error_info err;
  if (vpi_chk_error(&err)) {
    cout << "message '" << err.message << "', product '" << err.product <<
            "', code '" << err.code <<    "', file '" << err.file << "', line " << err.line << endl;
  }
}

// The callback function to be called after elaboration is finished
PLI_INT32 post_elab_callback(p_cb_data cb_data) {
    vpiHandle c { vpi_handle_by_name("b.c", NULL) };
    printErrorIfAny();
    vpi_printf("%s\n", vpi_get_str(vpiType, c));
    vpiHandle d { vpi_handle_by_name("b.c[0].inst.d", NULL) };
    printErrorIfAny();
    vpi_printf("%s\n", vpi_get_str(vpiType, d));
    return 0;
}

// The VPI registration function
void my_vpi_register() {
    s_cb_data cb_data;
    cb_data.reason = cbStartOfSimulation;
    cb_data.cb_rtn = post_elab_callback;
    cb_data.user_data = 0;
    vpi_register_cb(&cb_data);
}

// The VPI initialization function
extern "C" {
    void (*vlog_startup_routines[])() = {
        my_vpi_register,
        0
    };
}
```

marlonjames, 2023-05-03:
This might be the same issue as #3073.
There have been lots of issues related to generate support in the GPI code.