cocotb/cocotb #3296: error running example verilog using questa

richardc87, 2023-05-06:
### Discussed in https://github.com/cocotb/cocotb/discussions/3295

<div type='discussions-op-text'>

<sup>Originally posted by **richardc87** May  5, 2023</sup>
First time user.  I get this error message, and don't know how to proceed.  Thank you for your help.
**setup: miniconda conda ver 23.3.1, wiindows 10 64bit, questa 2020.2**

```
# Loading sv_std.std
# Loading work.count_up(fast)
# Loading C:/Users//AppData/Local/miniconda3/lib/site-packages/cocotb/libs/cocotbvpi_modelsim.dll
#      -.--ns ERROR    gpi                                ..s\cocotb_utils.cpp:56   in utils_dyn_open                  Unable to open lib C:\Users\\AppData\Local\miniconda3\python310.dll: The specified module could not be found.
#
#
#
#
#      -.--ns INFO     gpi                                ..\gpi\GpiCommon.cpp:101  in gpi_print_registered_impl       VPI registered
#
# End time: 10:17:11 on May 05,2023, Elapsed time: 0:00:04
# Errors: 0, Warnings: 1
ERROR: results.xml was not written by the simulation!
/c/Users//AppData/Local/miniconda3/lib/site-packages/cocotb/share/makefiles/simulators/Makefile.questa:158: recipe for target 'results.xml' failed
make[1]: *** [results.xml] Error 1
make[1]: Leaving directory '/c/test_project'
C:\Users\\AppData\Local\miniconda3\lib\site-packages\cocotb\share\makefiles/Makefile.inc:39: recipe for target 'sim' failed
make: *** [sim] Error 2
```
</div>

douwenqiangfpga, 2023-05-07:
I also encountered this problem, is the problem solved now?

marlonjames, 2023-05-07:
See #3295

closed 2023-05-07 by marlonjames