cocotb/cocotb #3314: Find a solution for sphinx domaintools bugs

imphil, 2023-05-22:
Sphinx domaintools (https://github.com/sphinx-contrib/domaintools) seems to be unmaintained and has at least two issues that we stumble across regularly:

* https://github.com/sphinx-contrib/domaintools/pull/13 (patch by @eric-wieser)
* https://github.com/sphinx-contrib/domaintools/pull/13

Since these issues are rather fundamental, I'm wondering if we're close to the only user of sphinx-domaintools? What would be our alternatives?

imphil, 2023-05-22:
Oh sorry @eric-wieser, I meant to link to https://github.com/sphinx-contrib/domaintools/pull/7. I edited the original issue description.

ktbarrett, 2023-06-21:
We could make our own Makefile Sphinx domain. The amount of code involved looks to be extremely small. As a first pass we could just use the existing domaintools and makedomain source code with the appropriate patches. If a repo is made in the org, I'll work on setting it up this weekend.

imphil, 2023-06-21:
Since it's probably a very small sphinx extension we can probably have it simply within the cocotb repository? Looking at domaintools and the makefile domain it seems that we're the only user.

imphil, 2023-06-21:
If you prefer, I can of course also setup a repo or repos within the cocotb GitHub org, let me know what which ones you'd like me to create.

ktbarrett, 2023-06-22:
I can do it locally.

PR #3352 (2023-07-05): Use vendored fork of domaintools
Closes #3314. Vendors in the [sphinxcontrib-domaintools](https://github.com/sphinx-contrib/domaintools) repo with the patches mentioned in https://github.com/cocotb/cocotb/issues/3314#issue-1719488035 applied. Unfortunately I had to add the dependency to the noxfile instead of the requirements file since relative paths in requirements.txt aren't handled very well (https://github.com/pypa/pip/issues/8765).
closed 2023-07-05 by imphil