cocotb/cocotb #3320: COCOTB_HDL_TIMEPRECISION not used in Questa VCOM

tj-scherer, 2023-05-26:
I was attempting to set the time precision to 1ps in Modelsim (which uses Questa makefile), with a top-level VHDL file. The two variables COCOTB_HDL_TIMEUNIT and COCOTB_HDL_TIMEPRECISION are only used in VLOG_OPTS which is used for verilog compilation. For VHDL, the time must be specified in VSIM_ARGS like `-t $(COCTB_HDL_TIMEPRECISION)`.

ktbarrett, 2023-05-31:
VLOG_ARGS only uses COCOTB_HDL_TIMEPRECISION if it isn't already defined. Should we also be unconditionally appending that? Perhaps instead we should only be appending it if it is defined, but ofc cover both VSIM_ARGS and VLOG_ARGS.

tj-scherer, 2023-06-02:
`vsim -help General` gives the following information:

```
# -t [1|10|100]fs|ps|ns|us|ms|sec Time resolution limit
#                                 (VHDL default: resolution setting from .ini
#                                 file)
#                                 (Verilog default: minimum time_precision in the
#                                 design)
```

`vlog -help Language` gives this:

```
-timescale[=]<timescale>        Specify the default timescale for modules not
                                having an explicit timescale. The format of
                                <timescale> is the same as that of the
                                `timescale directive. For example, -timescale
                                "1 ns / 1 ps".
```

I'm not familiar enough with Verilog compilation to say whether or not an unspecified timescale would cause problems (if not specified in the source or the vlog arg), and only in vsim. It appears to me that we should only use it in vsim, since verilog designs will just use the minimum specified in the design by default... not sure if using it in vlog actually makes a difference or not.

ktbarrett, 2023-06-06:
Alright. I guess the PR solves the problem. No need increase the scope.

tj-scherer, 2023-06-07:
Closed by #3321

closed 2023-06-07 by tj-scherer