cocotb/cocotb #3341: cocotb runner simulation script dump

vborchsh, 2023-06-15:
cocotb 1.8.0

What do you think about extending cocotb.runner for creating simulation script file?

There was such a feature in Makefile flow. It was useful for "hot" restart simulator withou closing the windows, reruning cocotb, opening windows again. In my case - for Modelsim.

It might something like dumpscript=True/False parameter. TBH, the only thing I need - is simulator restart without exercises. And mentioned script is just workaround, something like mentioned here: https://github.com/cocotb/cocotb/issues/284

Thanks.

vborchsh, 2023-06-16:
The similar thing already exist for Riviera, as far as I can see, but for tempfile

ktbarrett, 2023-06-22:
>There was such a feature in Makefile flow.

I wouldn't call it a feature, I'd call it a happy accident.