cocotb/cocotb #3354: Stack Trace shows scheduler code and not user code under some circumstances

jwprice100, 2023-06-29:
cocotb version: 1.8.2
operating system: Ubuntu 22.04 64 bit
Simulator: Riviera Pro 2022.10 64 bit
Python Version: 3.11.1 self install

Description: Under certain conditions an assertion failure results in a stack trace from scheduler code and not the user code that is typical. The exact conditions is two or more start_soon's in a row

I've attached an example using the new python runner. To run an example where the assertion failure output is correct run:
`TESTCASE=dummy_test_good SIM=riviera HDL_TOPLEVEL_LANG=vhdl pytest test_dummy.py`

and to run an example where the output is incorrect run:
`TESTCASE=dummy_test_bad SIM=riviera HDL_TOPLEVEL_LANG=vhdl pytest test_dummy.py`

Below is an example of the bad output:
`# COUT:   exec(co, module.__dict__)
# COUT:      0.00ns INFO     cocotb.regression                  Found test test_dummy.dummy_test_bad
# COUT:      0.00ns INFO     cocotb.regression                  Found test test_dummy.dummy_test_good
# COUT:      0.00ns INFO     cocotb.regression                  running dummy_test_bad (1/2)
# COUT: Traceback (most recent call last):
# COUT:   File "/home/jprice/aps_env/lib/python3.11/site-packages/cocotb/scheduler.py", line 633, in _resume_coro_upon
# COUT:     trigger.prime(self._react)
# COUT:   File "/home/jprice/aps_env/lib/python3.11/site-packages/cocotb/triggers.py", line 685, in prime
# COUT:     callback(self)
# COUT:   File "/home/jprice/aps_env/lib/python3.11/site-packages/cocotb/scheduler.py", line 384, in _react
# COUT:     self._event_loop(trigger)
# COUT:   File "/home/jprice/aps_env/lib/python3.11/site-packages/cocotb/scheduler.py", line 487, in _event_loop
# COUT:     self._schedule(coro, trigger=trigger)
# COUT:   File "/home/jprice/aps_env/lib/python3.11/site-packages/cocotb/scheduler.py", line 957, in _schedule
# COUT:     self._unschedule(coroutine)
# COUT:   File "/home/jprice/aps_env/lib/python3.11/site-packages/cocotb/scheduler.py", line 571, in _unschedule
# COUT:     self._cleanup()
# COUT:   File "/home/jprice/aps_env/lib/python3.11/site-packages/cocotb/scheduler.py", line 1094, in _cleanup
# COUT:     assert not self._pending_coros
# COUT: AssertionError
# COUT: 
# COUT: During handling of the above exception, another exception occurred:
# COUT: 
# COUT: Traceback (most recent call last):
# COUT:   File "/home/jprice/aps_env/lib/python3.11/site-packages/cocotb/__init__.py", line 233, in _initialise_testbench
# COUT:     _initialise_testbench_(argv_)
# COUT:   File "/home/jprice/aps_env/lib/python3.11/site-packages/cocotb/__init__.py", line 322, in _initialise_testbench_
# COUT:     regression_manager._execute()
# COUT:   File "/home/jprice/aps_env/lib/python3.11/site-packages/cocotb/regression.py", line 554, in _execute
# COUT:     return self._start_test()
# COUT:            ^^^^^^^^^^^^^^^^^^
# COUT:   File "/home/jprice/aps_env/lib/python3.11/site-packages/cocotb/regression.py", line 576, in _start_test
# COUT:     cocotb.scheduler._add_test(self._test_task)
# COUT:   File "/home/jprice/aps_env/lib/python3.11/site-packages/cocotb/scheduler.py", line 842, in _add_test
# COUT:     self._resume_coro_upon(
# COUT:   File "/home/jprice/aps_env/lib/python3.11/site-packages/cocotb/scheduler.py", line 636, in _resume_coro_upon
# COUT:     self._trigger2coros.pop(trigger)
# COUT: KeyError: <NullTrigger for Start <Test dummy_test_bad> at 0x7f8ea30afb50>
# COUT:      0.00ns ERROR    cocotb.scheduler                   Failing test at simulator request before test run completion: Simulator shut down prematurely
# COUT: sys:1: RuntimeWarning: coroutine 'dummy_coroutine' was never awaited
# KERNEL: Simulation has finished. There are no more test vectors to simulate.`
[example.zip](https://github.com/cocotb/cocotb/files/11901137/example.zip)

ktbarrett, 2023-07-02:
@jwprice100 Did you forget to attach the example?

jwprice100, 2023-07-03:
Whoops sorry here you go!

[example.zip](https://github.com/cocotb/cocotb/files/11936473/example.zip)

marlonjames, 2023-07-05:
The cause of this is for loop:
https://github.com/cocotb/cocotb/blob/08ef2f60f2816eda822ca953bc148fca40d2f506/cocotb/scheduler.py#L1091-L1094

`task.kill()` calls `Scheduler._unschedule()`, which removes the task from `Scheduler._pending_coros` while we are iterating over it.

ktbarrett, 2023-07-05:
I'm guessing we just need to change that to `while self._pending_coros`, then we can remove the assert?

ktbarrett, 2023-07-05:
That could pretty easily go into a 1.8.1.

jwprice100, 2023-07-05:
I would appreciate it if the fix could be incorporated. I love the Riviera fix in 1.8 but now whenever I start up a new testbench I get a million hard to trace exceptions. I'm happy to test it.

PR #3358 (2023-07-06): Scheduler: Fix _pending_coros cleanup
Closes #3354.

Previously, `_cleanup()` would iterate over `_pending_coros` in a for loop.
Because `Task.kill()` calls `_unschedule()`, the Task was removed from `_pending_coros` while iterating over it.
Use a while loop to ensure all of the Tasks are killed and removed.