cocotb/cocotb #3369: [Tracking] cocotb 1.8.1

ktbarrett, 2023-07-13:
Cherry-pick the following PRs.

- [ ] #3358
- [ ] #3361

imphil, 2023-09-14:
@cocotb/maintainers Is there anything else we want for 1.8.1 in terms of high-value bug fixes?

I'm wondering if we should do a 1.9 soon as well to include all the other valuable (small-ish) fixes that went into master, I think we only have one breaking fix (the removal of the public decorator) that we cannot have in a 1.x release?

rw1nkler, 2023-09-21:
May I ask if there is any information available regarding the potential release date of Cocotb 1.8.1?

imphil, 2023-10-02:
The last batch of cherry-picks: https://github.com/cocotb/cocotb/pull/3429

imphil, 2023-10-02:
@rw1nkler: It should be available any day now.

imphil, 2023-10-07:
https://github.com/cocotb/cocotb/releases/tag/v1.8.1 is now out. Enjoy!

closed 2023-10-07 by imphil