cocotb/cocotb #3372: Xcelium: writes to multi-dimensional arrays are partially ignored

imphil, 2023-07-15:
### Description
Some writes to multi-dimensional arrays in Xcelium 23.03-s004 never make it into the design.

The following two signals are defined in a SystemVerilog design:

https://github.com/cocotb/cocotb/blob/5d463f7ecd017efa0b15448ace8641e93aef7d0d/tests/designs/array_module/array_module.sv#L82-L85

These signals are then written in a cocotb test:

https://github.com/cocotb/cocotb/blob/5d463f7ecd017efa0b15448ace8641e93aef7d0d/tests/test_cases/test_array/test_array.py#L130-L136

And later on these signals are read.

### Expected result
All elements of the array are set.

https://github.com/cocotb/cocotb/blob/5d463f7ecd017efa0b15448ace8641e93aef7d0d/tests/test_cases/test_array/test_array.py#L162-L166

There are no simulator-specific overrides in place for this test, i.e. we get the expected outcome for all simulators (except GHDL, where the full test is disabled).

### Actual result

See waveform: It seems that only the lowest element is set, all the others remain unaffected by the write.

[image]

### Simulator
Xcelium 23.03-s004

s3bs, 2024-07-07:
If the signals are changed to `logic` 
```
logic [0:3]       sig_t1;
logic [7:0]       sig_t2[7:4];
logic [7:0]       sig_t3a[1:4];
logic [7:0]       sig_t3b[3:0];
logic [7:0]       sig_t4[0:3][7:4];
logic [7:0]       sig_t5[0:2][0:3];
logic [7:0]       sig_t6[0:1][2:4];
```
this passes. 
I've encountered the same for module ports.

s3bs, 2024-07-16:
just had the chance to test this on the latest release `Xcelium 24.03-s004` and it works out of the box. using cocotb v1.8.0
`COCOTB_LOG_LEVEL=TRACE TESTCASE=test_read_write make -C tests/test_cases/test_array/ SIM=xcelium` passes now

PR #4046 (2024-07-16): Renable checks in test_array following upstream fixes to Xcelium
Closes #3372.
closed 2024-07-16 by ktbarrett