cocotb/cocotb #3377: How to check the signal value inside a for loop

firemountain154B, 2023-07-18:
When I'm trying to grap a instance in the for loop, it seems I cannot do that, and there is not any template.

like if define a for loop in system verilog
```
for (genvar i=0; i<NUM; i++): loop1
  logic test;
  assign test = data_in;
```

in the testbench part, I cannot grab the test directly using 
```
print(dut.loop1[0].test.value)
```
it returns bug information "no object named loop1 in module".

So how can I do that.
Or it's not support in cocotb

ktbarrett, 2023-07-18:
This should be possible. It may be due to limited support in your simulator. Please provide the full details about your environment that the issue template asked you.

genfrontier, 2023-07-26:
Hey guys, sorry to intrude on firemountain issue , just to add that i see the same issue albeit on a more convoluted fashion but it boils down to the same. I am using 1.7.2 cocotb, using VCS: VCS Release version S-2021.09-SP2-1_Full64 and here is a snippet: `

genvar stream_group_gen;
genvar stream_gen;

for (stream_group_gen = 0; stream_group_gen < 8; stream_group_gen++) begin: overlay_stream_group

for (stream_gen = 0; stream_gen < 8; stream_gen++) begin: overlay_stream

stream_module stream_inst()

end
end
`

When i try to access this via : `

dut.overlay_stream_group[0].overlay_stream[0].stream_inst"

`

I get the same error as the OP, a print(dir(dut)) gives no member named overlay_stream_group, but weirdly we get a member named stream_inst. which is not an array so any access to it while it does not complain at runtime , goes to nowhere

genfrontier, 2023-07-26:
Thanks ktbarret, i was able to bypass this by using : 
      
      fullsig = dut._id(f"overlay_stream_group[{groupsel:d}].overlay_stream[{streamsel:d}].stream_inst",extended=False)
      
      The trick is to use as the ._id string the whole hierarchy from the first generate loop onward. Leaving it here as might someone find it helpful

closed 2023-09-03 by github-actions[bot]

firemountain154B, 2023-09-06:
may I have a elaborate explanation on what the mean of simulator I use, I guess it's verilator? and I just built it with python3, I still have this problem, so how u grab this signal if it's defined like that.

ktbarrett, 2023-09-10:
Verilator is a simulator. What version of verilator? You can find out by running `verilator --version`.