cocotb/cocotb #3384: excessive memory consumption with Modelsim

yakovlevyury, 2023-07-28:
Hello, I have problem with memory consumption when run cocotb tests with Modelsim, problem same as in #466 .  With the same minimal example of problem i have about 5 MB/s memory leak. This dont allow to run tests for long time, my 32 GB of RAM overflows in about 4-5 hours of simulation.
Ubuntu Linux 18.04 64-bit
cocotb 1.8.0
Modelsim SE-64 2020.4 Revision 2020.10
python 3.8.16

```
import cocotb
from cocotb.triggers import Timer
from cocotb.triggers import RisingEdge

@cocotb.test()
async def ctbml(dut):
	dut._log.info("ctbml start")
	cocotb.start_soon(clock_100mhz(dut.clock))

	dut.reset.value = 1
	await RisingEdge(dut.clock)
	dut.reset.value = 0
	await RisingEdge(dut.clock)

	await Timer(10, units="sec")
	dut._log.info("ctbml finished")

async def clock_100mhz(signal):
	period = 10000
	while True:
		signal.value <= 1
		await Timer(period / 2, units = "ps")
		signal.value <= 0
		await Timer(period / 2, units = "ps")
```

```
`timescale 1ns / 1ns

module top_ctbml(
input clock,
input reset,
output reg [23:0] test
);

initial test = 0;

always @(posedge clock)
begin
    if(reset) test <= 0;
    else test <= test + 1;
end

endmodule
```

```
# General cocotb settings
exec cocotb-config --prefix > /tmp/cocotb-prefix
set COCOTB_DIR [string trim [read [open "/tmp/cocotb-prefix" r]]]
set VPI_LIB $COCOTB_DIR/cocotb/libs/libcocotbvpi_modelsim.so
set ::env(COCOTB_SIM) 1
set ::env(WAVES) 1
set ::env(SIM) modelsim
# set ::env(ARCH) i686
set ::env(GUI) 1
set ::env(PYTHONPATH) ${COCOTB_DIR}/cocotb/libs/modelsim

# This test settings
set ::env(MODULE) test_design
set ::env(PYTHONPATH) ./:../model:$::env(PYTHONPATH)

vlog ./top_ctmbl.v

vsim -voptargs=+acc -pli $VPI_LIB work.top_ctbml -onfinish stop
```

```
(base) yakovlev@yakovlev:~$ ps aux | grep vsim ; date
yakovlev 31378 15.0  0.3 319488 103044 pts/17  S+   14:36   0:02 /home/yakovlev/hdd_wd/modelsim/modeltech/bin/../linux_x86_64/vish -- -vsim -gui -do run_cocotb.tcl
yakovlev 31455 41.0  0.3 368044 111600 pts/17  R+   14:36   0:06 /home/yakovlev/hdd_wd/modelsim/modeltech/linux_x86_64/vsimk -port 127.0.0.1:37867 -stdoutfilename /tmp/VSOUTSwvoWK -gui -do run_cocotb.tcl
yakovlev 31564  0.0  0.0  15664  1168 pts/16   S+   14:36   0:00 grep --color=auto vsim
Fri 28 Jul 14:36:43 +05 2023
(base) yakovlev@yakovlev:~$ ps aux | grep vsim ; date
yakovlev 31378 11.0  0.3 319488 103044 pts/17  S+   14:36   0:02 /home/yakovlev/hdd_wd/modelsim/modeltech/bin/../linux_x86_64/vish -- -vsim -gui -do run_cocotb.tcl
yakovlev 31455 60.8  0.4 368044 154104 pts/17  R+   14:36   0:13 /home/yakovlev/hdd_wd/modelsim/modeltech/linux_x86_64/vsimk -port 127.0.0.1:37867 -stdoutfilename /tmp/VSOUTSwvoWK -gui -do run_cocotb.tcl
yakovlev 31601  0.0  0.0  15664  1080 pts/16   S+   14:36   0:00 grep --color=auto vsim
Fri 28 Jul 14:36:50 +05 2023
(base) yakovlev@yakovlev:~$ ps aux | grep vsim ; date
yakovlev 31378  7.5  0.3 319488 103044 pts/17  S+   14:36   0:02 /home/yakovlev/hdd_wd/modelsim/modeltech/bin/../linux_x86_64/vish -- -vsim -gui -do run_cocotb.tcl
yakovlev 31455 75.2  0.7 499116 230928 pts/17  R+   14:36   0:26 /home/yakovlev/hdd_wd/modelsim/modeltech/linux_x86_64/vsimk -port 127.0.0.1:37867 -stdoutfilename /tmp/VSOUTSwvoWK -gui -do run_cocotb.tcl
yakovlev 31633  0.0  0.0  15664  1032 pts/16   S+   14:37   0:00 grep --color=auto vsim
Fri 28 Jul 14:37:03 +05 2023
(base) yakovlev@yakovlev:~$ ps aux | grep vsim ; date
yakovlev 31378  6.7  0.3 319488 103044 pts/17  S+   14:36   0:02 /home/yakovlev/hdd_wd/modelsim/modeltech/bin/../linux_x86_64/vish -- -vsim -gui -do run_cocotb.tcl
yakovlev 31455 77.5  0.7 499116 258384 pts/17  R+   14:36   0:31 /home/yakovlev/hdd_wd/modelsim/modeltech/linux_x86_64/vsimk -port 127.0.0.1:37867 -stdoutfilename /tmp/VSOUTSwvoWK -gui -do run_cocotb.tcl
yakovlev 31646  0.0  0.0  15664  1136 pts/16   S+   14:37   0:00 grep --color=auto vsim
Fri 28 Jul 14:37:08 +05 2023
(base) yakovlev@yakovlev:~$ 

```

marlonjames, 2023-08-02:
Have you tried running the simulation with valgrind to find the source of the memory leak?

yakovlevyury, 2023-08-03:
> Have you tried running the simulation with valgrind to find the source of the memory leak?

No, from #466 and other discussion in https://www.linux.org.ru/forum/development/13834298 (valgrind was one of options in this discussion, there were some problems with connect valgrind to modelsim)  I was some frightened about difficulty to connect debugger to cocotb+modelsim.  I will try in few days to debug this, but i have no experience in work with valgrind and findind memory leaks yet, it will take some time...)