cocotb/cocotb #3394: Issues on running GHDL with cocotb .. VPI-related

lukipedio, 2023-08-10:
When running an example test from cocotb master repo (test_addr.py) I get the following error message on windows 11:

C:\work\cocotb-master\examples\adder\tests\test_adder.py:10: UserWarning: Python runners and associated APIs are an experimental feature and subject to change.
  from cocotb.runner import get_runner
INFO: Running command ghdl -i --work=top 'C:\work\cocotb-master\examples\adder\hdl\adder.vhdl' in directory C:\work\cocotb-master\examples\adder\tests\sim_build
INFO: Running command ghdl -m --work=top adder in directory C:\work\cocotb-master\examples\adder\tests\sim_build
INFO: Running command ghdl -r --work=top adder --vpi=C:/Users/lcolombini/AppData/Local/Programs/Python/Python311/Lib/site-packages/cocotb/libs/cocotbvpi_ghdl.dll in directory C:\work\cocotb-master\examples\adder\tests\sim_build
loading VPI module 'C:/Users/lcolombini/AppData/Local/Programs/Python/Python311/Lib/site-packages/cocotb/libs/cocotbvpi_ghdl.dll'
Impossibile trovare il modulo specificato.

C:\tools\ghdl\bin\ghdl.exe:error: cannot load VPI module 'C:/Users/lcolombini/AppData/Local/Programs/Python/Python311/Lib/site-packages/cocotb/libs/cocotbvpi_ghdl.dll'
Process 'ghdl' terminated with error 1

It complains about a DLL from cocotb (cocotbvpi_ghdl.dll).

The GHDL has been extracted from https://github.com/ghdl/ghdl/releases/download/v3.0.0/ghdl-MINGW32.zip

themperek, 2023-08-10:
This is a 32-bit version. You need 32-bit python to run.

lukipedio, 2023-08-11:
I have switched to Python 3.11 32-bit, as you can see, but still getting this error message:

PS C:\work\cocotb-master\examples\adder\tests> python .\test_adder.py
C:\work\cocotb-master\examples\adder\tests\test_adder.py:10: UserWarning: Python runners and associated APIs are an experimental feature and subject to change.
  from cocotb.runner import get_runner
INFO: Running command ghdl -i --work=top 'C:\work\cocotb-master\examples\adder\hdl\adder.vhdl' in directory C:\work\cocotb-master\examples\adder\tests\sim_build
INFO: Running command ghdl -m --work=top adder in directory C:\work\cocotb-master\examples\adder\tests\sim_build
INFO: Running command ghdl -r --work=top adder --vpi=C:/Users/lcolombini/AppData/Local/Programs/Python/Python311-32/Lib/site-packages/cocotb/libs/cocotbvpi_ghdl.dll in directory C:\work\cocotb-master\examples\adder\tests\sim_build
loading VPI module 'C:/Users/lcolombini/AppData/Local/Programs/Python/Python311-32/Lib/site-packages/cocotb/libs/cocotbvpi_ghdl.dll'
Impossibile trovare il modulo specificato.

C:\tools\ghdl\bin\ghdl.exe:error: cannot load VPI module 'C:/Users/lcolombini/AppData/Local/Programs/Python/Python311-32/Lib/site-packages/cocotb/libs/cocotbvpi_ghdl.dll'
Process 'ghdl' terminated with error 1

I think I'll switch to work on a dockerized version.

lukipedio, 2023-08-11:
I think it is related to https://github.com/cocotb/cocotb/issues/2030

lukipedio, 2023-08-11:
Yes, both WSL and Docker work just fine.

closed 2023-08-11 by lukipedio