cocotb/cocotb #3402: I have got cocotb,but i have got a problem?

OldTomCrazyCode, 2023-08-30:


OldTomCrazyCode, 2023-08-30:
[root@TTOOMM /home/root/ic_prjs/cocotb_test/cocotb/examples/simple_dff]$cocotb-config
usage: cocotb-config [-h] [--prefix] [--share] [--makefiles] [--python-bin] [--help-vars] [--libpython] [--lib-dir] [--lib-name INTERFACE SIMULATOR] [--lib-name-path INTERFACE SIMULATOR] [-v]

options:
  -h, --help            show this help message and exit
  --prefix              echo the package-prefix of cocotb
  --share               echo the package-share of cocotb
  --makefiles           echo the package-makefiles of cocotb
  --python-bin          echo the path to the Python binary cocotb is installed for
  --help-vars           show help about supported variables
  --libpython           Print the absolute path to the libpython associated with the current Python installation
  --lib-dir             Print the absolute path to the interface libraries location
  --lib-name INTERFACE SIMULATOR
                        Print the name of interface library for given interface (VPI/VHPI/FLI) and simulator
  --lib-name-path INTERFACE SIMULATOR
                        Print the absolute path of interface library for given interface (VPI/VHPI/FLI) and simulator
  -v, --version         echo the version of cocotb
[root@TTOOMM /home/root/ic_prjs/cocotb_test/cocotb/examples/simple_dff]$cocotb-config --lib-dir
/home/lib/python3.11/site-packages/cocotb/libs
[root@TTOOMM /home/root/ic_prjs/cocotb_test/cocotb/examples/simple_dff]$cocotb-config --libpython
[root@TTOOMM /home/root/ic_prjs/cocotb_test/cocotb/examples/simple_dff]$

closed 2023-08-30 by OldTomCrazyCode

OldTomCrazyCode, 2023-08-30:
I can't find the libpython of cocotb.....,   I will get a error "
/home/lib/python3.11/site-packages/cocotb/share/makefiles/Makefile.inc:159: *** 

find_libpython was not able to find a libpython in the current Python environment. Ensure
the Python development packages are installed. If they are installed and find_libpython
is not finding the path to libpython, file an upstream bug in find_libpython; then
manually override the LIBPYTHON_LOC make variable with the absolute path to libpython.so
(or python.dll on Windows)."
 
when i run the "make SIM=icarus".

why ???

just the libpython  can not be finded.

kaleb, 2023-08-30:
Thanks for mentioning me but I don't know if I can assist with this.

xmy01, 2024-04-29:
hi   I meet the same problem, your problem have been solved yet? thanks very much

ktbarrett, 2024-04-29:
@xmy01 Is `libpython3.X.so` available on your system? You may need to install the `libpython3.X-dev` package if you are on Debian/Ubuntu. If you are on another OS, please research how to install that library. If you are sure there's a `libpython3.X.so` available on your system and `cocotb-config --libpython` still returns nothing, open a ticket against [`find_libpython`](https://github.com/ktbarrett/find_libpython).