cocotb/cocotb #3403: I have got cocotb,but i can not find the path of libpython?

OldTomCrazyCode, 2023-08-30:
[root@TTOOMM /home/root/ic_prjs/cocotb_test/cocotb/examples/simple_dff]$cocotb-config
usage: cocotb-config [-h] [--prefix] [--share] [--makefiles] [--python-bin] [--help-vars] [--libpython] [--lib-dir] [--lib-name INTERFACE SIMULATOR] [--lib-name-path INTERFACE SIMULATOR] [-v]

options:
-h, --help show this help message and exit
--prefix echo the package-prefix of cocotb
--share echo the package-share of cocotb
--makefiles echo the package-makefiles of cocotb
--python-bin echo the path to the Python binary cocotb is installed for
--help-vars show help about supported variables
--libpython Print the absolute path to the libpython associated with the current Python installation
--lib-dir Print the absolute path to the interface libraries location
--lib-name INTERFACE SIMULATOR
Print the name of interface library for given interface (VPI/VHPI/FLI) and simulator
--lib-name-path INTERFACE SIMULATOR
Print the absolute path of interface library for given interface (VPI/VHPI/FLI) and simulator
-v, --version echo the version of cocotb
[root@TTOOMM /home/root/ic_prjs/cocotb_test/cocotb/examples/simple_dff]$cocotb-config --lib-dir
/home/lib/python3.11/site-packages/cocotb/libs
[root@TTOOMM /home/root/ic_prjs/cocotb_test/cocotb/examples/simple_dff]$cocotb-config --libpython
[root@TTOOMM /home/root/ic_prjs/cocotb_test/cocotb/examples/simple_dff]$

I can't find the libpython of cocotb....., I will get a error "
/home/lib/python3.11/site-packages/cocotb/share/makefiles/Makefile.inc:159: ***

find_libpython was not able to find a libpython in the current Python environment. Ensure
the Python development packages are installed. If they are installed and find_libpython
is not finding the path to libpython, file an upstream bug in find_libpython; then
manually override the LIBPYTHON_LOC make variable with the absolute path to libpython.so
(or python.dll on Windows)."

when i run the "make SIM=icarus".

why ???

just the libpython can not be finded.

please....
@nickg @ddribin @kaleb Barrett

imphil, 2023-08-30:
Hi @OldTomCrazyCode, welcome to cocotb!

We need to know a bit more about your environment: 
* What cocotb version are you using? 
* Which operating system are you using?
* How did you install Python?
* The output of `find_libpython --list-all` and `find_libpython --candidate-paths`.

In all cases, be as specific as possible.

OldTomCrazyCode, 2023-08-31:
Thanks for your reply, I had fixed the envirnmont error after installing the python again. Really thank you again!!

closed 2023-08-31 by OldTomCrazyCode

ktbarrett, 2024-04-17:
```
find_libpython was not able to find a libpython in the current Python environment. Ensure
the Python development packages are installed. If they are installed and find_libpython
is not finding the path to libpython, file an upstream bug in find_libpython; then
manually override the LIBPYTHON_LOC make variable with the absolute path to libpython.so (or python.dll on Windows).
```

Have you considered following the instructions here?

ktbarrett, 2024-04-17:
Are you certain that there is a `libpython3.X.so` or `python3.X.dll` installed on your system?
If there is and `find_libpython` is not finding it, open an issue in [the find_libpython project](https://github.com/ktbarrett/find_libpython).

ktbarrett, 2024-04-17:
Also, don't bump old support issues, make a new one. And please fill out the full set of information about your environment; it's difficult to impossible to help otherwise.

mounikakallakola1, 2024-04-17:
Sorry, i donot see any libpython.so installed in my virtual environment.
I will create a new issue. Thanks