cocotb/cocotb #3404: Mixed-languages multi-libraries not working as expected

oliverbm67, 2023-09-01:
# Issue description #
compilation of a design containing both Verilog and VHDL and also mixing libraries does not work. The RTL and the python script can be found here : https://github.com/oliverbm67/cocotb/tree/e63dda34cbfd64b0600cbbb3bfe4e1db7f26707c/examples/mixed_language_multi_libraries

I noticed the following :

- If hdl_toplevel is not provided in the runner.build a default top is used in the tool command
- If no hdl_library is provided, cocotb 

does not default to work as (at least in my experience) most tools do

# Logs #
The run.do script used to check compilation with questa is also attached
[cocotb.log](https://github.com/cocotb/cocotb/files/12500962/cocotb.log)
[modelsim.log](https://github.com/cocotb/cocotb/files/12500976/modelsim.log)
[run.do.txt](https://github.com/cocotb/cocotb/files/12500980/run.do.txt)

# Environment #
Python 3.11.4
cocotb 1.8.0
Questa 2021.2 ( from Intel FPGA suite)

# Notes #
I do think using Xcelium gives similar results.

marlonjames, 2023-09-03:
The Python runner is experimental and targeted to cocotb regression tests. You are free to copy and modify for your use, or [integrate cocotb](https://docs.cocotb.org/en/stable/custom_flows.html) to your build / simulate workflow.

oliverbm67, 2023-09-03:
I tried the python runner because I could not figure out how to do it with the standard Makefile flow. I updated my example with a cocotb Makefile ( https://github.com/oliverbm67/cocotb/blob/63ecde3f9527a8d2374e349a651ea82e1724d2a8/examples/mixed_language_multi_libraries/cocotb/Makefile ), but it still has compilation order and library management issue.

cmarqu, 2023-09-04:
FWIW, #1707 is a relevant issue regarding the Makefiles.

ktbarrett, 2023-09-06:
I don't see the problem? `work` is not a legal library name as of VHDL 2002, we aren't going to default to that. The build of the `reg_b` sources is not associated with a library and is called out in `top` as being in `cell_lib`, so of course it won't compile correctly. 

You've pasted a bunch of stuff and said "it doesn't work", but you haven't explained what you expect the behavior to be, why, and what needs to be changed to make it work.

Based on your post, I can reason that you think that not specifying a library name when building a library should put the entities into a default library. IMO, this argument should instead be required. Building libraries are different than building a top-level design. I wanted them to be different methods originally, but that isn't what ended up happening. Instead, we need to force the two distinct methods into one and reject invalid calls.

cmarqu, 2023-09-06:
Nothing prevents us from separating the methods though, we always declared the Python runner as experimental.

oliverbm67, 2023-09-06:
Sorry for being unclear with the initial question. I was expecting cocotb compilation to successfully compile the initial code as it was possible to do so with what I thought was an equivalent modelsim compilation script. I think in this case Modelsim is too permissive and/or too forgiving.

I have updated my example according to your feedback to give explicit libraries in all cases except the top level. Using the python runner works in this latest example, but there is still an issue with the Makefile flow. 

There seem to be 2 issues at play :

1. All VHDL code is compiled first, including top level. Thus the needed Verilog are not compiled in the correct order
2. It does not seem possible to specify a compilation library for verilog files ( I would expect VERILOG_SOURCES_mylib to work ?)

oliverbm67, 2023-11-14:
From the original question I take the following :

- Complicated compilation setup can be handled using the python runner, but as this is considered beta feature it is not recommended
- Makefile flow do not support this type of use case as is. But this is already tracked by https://github.com/cocotb/cocotb/issues/1707 (at least partially)

I can make a pull request with the example code if it can be useful for testing or demo, otherwise the issue can be closed

ktbarrett, 2023-11-30:
> There seem to be 2 issues at play :
> 
> 1. All VHDL code is compiled first, including top level. Thus the needed Verilog are not compiled in the correct order
> 2. It does not seem possible to specify a compilation library for verilog files ( I would expect VERILOG_SOURCES_mylib to work ?)

These are both valid criticisms and limitations. Separating the VHDL and Verilog sources into separate variables was never a good idea. Mixed language simulators do allow creating libraries with only verilog code (and this is even supported by the SV standard), but the Makefiles weren't designed with them in mind and I don't think we will both updating them. However, I agree this should be solved by the Python runner.

Nabih-Saleh, 2024-02-27:
It is very frustrating that we run in a loop, the same problem but described differently in tens of tickets and none of them yet solved.

ktbarrett, 2024-02-27:
@Nabih-Saleh You are free to fix it yourself or pay someone to. If not, you'll just have to wait for someone else.

PR #3796 (2024-04-10): Introduce language-agnostic `sources` arg to runner
Closes #3404. Introduces a language-agnostic `sources` argument to the runner. These are segregated into Verilog and VHDL sources when necessary by using the extension, or new tagging types `VHDL` and `Verilog`. The tagging types can also be applied to `build_args`.

- [x] newsfragment
- [x] documentation
- [x] tests
closed 2024-04-10 by ktbarrett