cocotb/cocotb #3406: How to check the signal inside the loop

firemountain154B, 2023-09-06:
Hi I'm coming back again, sorry for not subscribe this before, so this issue before was closed, but I still have such a problem.
The simulator is verilator, and environment is like 
```
import cocotb
from cocotb.triggers import Timer
from cocotb.triggers import FallingEdge
from cocotb.clock import Clock
from cocotb.runner import get_runner
```
When I'm trying to grap a instance in the for loop, it seems I cannot do that, and there is not any template.

like if define a for loop in system verilog
```
for (genvar i=0; i<NUM; i++): loop1
  logic test;
  assign test = data_in;
```

in the testbench part, I cannot grab the test directly using 
```
print(dut.loop1[0].test.value)
```
it returns bug information "no object named loop1 in module".

So how can I do that.
Or it's not support in cocotb

ktbarrett, 2023-09-10:
That is supported. You need to give more information or we can't help you. Post the version of verilator you are using and an example project that has the issue, including the cocotb test file, your HDL files, and the project Makefile.

firemountain154B, 2023-09-23:
For example: if I just write a test hdl files as following
```
`timescale 1ns / 1ps
module test (
    input clk, 
    input rst,
    input [3:0] in_1[3:0],
    output [3:0] out
);

for (genvar i=0; i<4; i++) begin: for_test
    logic [3:0] test;
    assign test = in_1[i];
end
    
assign out = for_test[3].test;
endmodule

```

The testbench file
```
#!/usr/bin/env python3

# This script tests the register slice
import os

import cocotb
from cocotb.triggers import Timer
from cocotb.triggers import FallingEdge
from cocotb.clock import Clock
from cocotb.runner import get_runner

@cocotb.test()
async def test_register_slice(dut):
    """Test register slice"""
    samples = 20

    # Reset cycle
    await Timer(20, units="ns")
    dut.rst.value = 1
    await Timer(100, units="ns")
    dut.rst.value = 0

    # Create a 10ns-period clock on port clk
    clock = Clock(dut.clk, 10, units="ns")
    # Start the clock
    cocotb.start_soon(clock.start())
    await Timer(500, units="ns")

    for i in range(100):
        await FallingEdge(dut.clk)
        breakpoint()
        dut.in_1.value = [8,7,6,5]

def runner():
    sim = os.getenv("SIM", "verilator")

    verilog_sources = [
        "../../../../components/ViT/test.sv",
    ]
    runner = get_runner(sim)
    runner.build(
        verilog_sources=verilog_sources,
        hdl_toplevel="test",
    )
    runner.test(hdl_toplevel="test", test_module="test_tb")

if __name__ == "__main__":
    runner()

```

Just run the code `python3 test_tb.py`

When I'm trying to debug it with pdb, I want to check the signal inside the for loop.
The bug information is like
```
(Pdb) dut.in_1.value
[1000, 0111, 0110, 0101]
(Pdb) dut.out.value
1000
(Pdb) dut.for_test.value
*** AttributeError: test contains no object named for_test
(Pdb) dut.for_test[0].test.value
*** AttributeError: test contains no object named for_test
```

From this example, we can find that it can pass the signal inside for loop out.
But if I want to check it from testbench, I cannot get any result but the bug information.

Not only in Pdb, when I was trying to print it out, I also encountered this issue.

marlonjames, 2023-10-24:
Duplicate of #1884

marlonjames, 2023-10-24:
Unfortunately, Verilator's VPI support for generate constructs is incomplete.

closed 2023-10-31 by marlonjames