cocotb/cocotb #3418: LogicArray: Make type conversions more explicit

imphil, 2023-09-18:
Looks like we weren't super clear about `LogicArray` being "data storage only", with no interpretation of the stored data. I had a look at the API and I think we can be clearer there as well.

Right now we have the following API in `LogicArray`, in line with what we had in `BinaryValue`:

* Interpret as signed integer: `LogicArray.signed_integer`
* Interpret as unsigned integer: `LogicArray.integer`

Comments about this API:

* It is confusing, because C/C++ would say an `int` is *signed*, while we say `integer` (without further qualification) is *unsigned*.
* I find properties slightly out of place here for two reasons:
  * We're operating on the LogicArray data and convert/interpret it. To me that's an action (and hence a method), not a property.
  * Passing additional arguments to the conversion, e.g. the byte order, requires a method.

To make things very explicit, I would go for:
* Interpret as integer: `LogicArray.as_int(byteorder='big', signed=False)`

Properties of this proposed API:
* This API matches closely the arguments of [`int.from_bytes()`](https://docs.python.org/3/library/stdtypes.html#int.from_bytes) (which is similar in semantics)
* Using a method adds the ability to specify the byte order.
* We keep the naming we (and Python) had before: plain "integer" means "unsigned integer".
* We can easily extend this scheme for conversions to other data types, e.g., floats.

WDYT @ktbarrett and @cocotb/maintainers?

ktbarrett, 2023-09-18:
I'm fine with removing these and the proposed replacement. The only reason that `signed_integer` and `integer` were added was to maintain some amount of interface compatibility with BinaryValue to make transitioning easier, but you are correct in that there is an assumption of endianness that isn't a good idea.

aubindetrez, 2023-09-18:
I'm fine with implementing this feature as part of https://github.com/cocotb/cocotb/pull/3244.
I like the `LogicArray.as_int(byteorder='big', signed=False)` syntax and also the idea to extent it with float support... (would be better in a dedicated PR though)

ktbarrett, 2024-03-19:
So what do we need to support for conversions?

I'm thinking we need to support conversion to and from `int`s as if they were bitvectors and also as if they were endian-aware byte vectors. We can support the endian-aware byte vector construction by leveraging the existing `int.from_bytes` and `int().to_bytes()` calls by adding support for construction from `bytes`.

So add a minimum I suggest adding:
```python
class LogicArray:
    ...

    @classmethod
    def from_unsigned(cls, value: int, range: Range) -> LogicArray: ...

    @classmethod
    def from_signed(cls, value: int, range: Range) -> LogicArray: ...
    
    @classmethod
    def from_bytes(cls, value: bytes, range: Optional[Range] = None) -> LogicArray: ...
    
    def to_unsigned(self) -> int: ...
    
    def to_signed(self) -> int: ...
    
    def to_bytes(self) -> bytes: ...

    ...
```

Not supplying the `range` in `from_bytes` would infer the range as `len(b)*8-1 downto 0`.

ktbarrett, 2024-03-19:
Actually, `bytes` is kinda difficult to reason about. What bytes get mapped where? What are "successive memory locations"? Is it on the right or the left?

PR #3792 (2024-04-07): Cleanup LogicArray interface
Closes #3660. Closes #3418.

* Deprecate `LogicArray.integer`, `.signed_integer`, and `.binstr`
* Deprecate constructing `LogicArray` from ints.
* Introduce `LogicArray.from_unsigned`, `.from_signed`, `.to_signed`, and `.to_unsigned` to replace what was deprecated.

If we need other kinds of construction/conversion, we can add them later.

- [x] Add newsfragment.
closed 2024-04-07 by ktbarrett