cocotb/cocotb #3419: Xcelium/VHDL: Unable to get correct time precision

imphil, 2023-09-20:
Getting the simulator's precision in Xcelium with VHDL toplevels is impossible currently for two reasons:

1. Xcelium does not implement the VHPI call to get the precision in VDHL:  `vhpi_get_phys(vhpiResolutionLimitP, NULL);`, returning invalid data (high == -1, low == 0).
2. With Xcelium, [we always load VPI first](https://github.com/cocotb/cocotb/blob/1749b85dfb7898db9096d00ee31ec5369c3f5ff6/cocotb/share/makefiles/simulators/Makefile.xcelium#L84) and then VHPI as second GPI implementation. However, the `cocotb.simulator.get_precision()` call [only queries the first GPI implementation](https://github.com/cocotb/cocotb/blob/1749b85dfb7898db9096d00ee31ec5369c3f5ff6/cocotb/share/lib/gpi/GpiCommon.cpp#L217), which is always VPI. 
3. The VPI time precision seems to be wrong: For unknown reasons (yet), whatever you set as `-timescale` to xrun, a precision of 1 ns (-9) is returned.

So there are multiple things to sort out for us and Cadence. I opened a Cadence case to ask for support for `vhpi_get_phys()`, but that alone won't do the trick. Ideally, we want to load VPI only if Verilog is used. We'll need to look into what's preventing that.

tymonx, 2024-08-28:
I have workaround that limitation by overriding the Cocotb internal function `_get_simulation_precision`. Tested with Cocotb `1.9.0` and Cadence Xcelium `24.03.005`. It works perfectly fine with VHDL as top level design. I can confirm that the `vhpi_get_phys` is not implemented and VPI version is not working correctly (it returns `-9`). But there is no problem to generate signals like clocks with precision below `1ns` using VHPI. Maximum value is `-15` (`fs` resolution) according to VHDL specification and Cadence Xcelium prints time with `fs` resolution.

Example workaround:

```python
import cocotb

cocotb.utils._get_simulator_precision = lambda: -12

#  Cocotb tests
```

tymonx, 2025-03-15:
@imphil I have found clean workaround:

1. Create a dummy Verilog/SystemVerilog module:

```verilog
module cocotb;
endmodule
```

2. Compile it normally to any library. Could be also default `work` library. It doesn't matter.
3. Add compiled dummy module as additional top level design `-top cocotb`. Cadence Xcelium supports multi `-top` invocations.
4. Time precision is working as expected but through VPI
5. Profit 🚀 

Tested with Cadence Xcelium `24.03.005` and `24.09.006`. If someone is working with Xilinx Vivado then some their primitives and IPs are requiring to add `-top glbl` anyway because it generates `100ns` global reset for other modules. `glbl` is written in Verilog.