cocotb/cocotb #3421: VHPI: Registered callback isn't enabled!

henrikmpedersen00, 2023-09-21:
**Environment:**
Linux  5.15.90.1-microsoft-standard-WSL2 #1 SMP Fri Jan 27 02:56:13 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux
Python 3.8.10 (default, May 26 2023, 14:05:08)
[GCC 9.4.0] on linux
TOOL:   xrun    21.03-s015: Exiting on Sep 21, 2023 at 15:35:39 +04  (total: 00:00:00) 64bit version
cocotb in /home/henrikmpedersen/.local/lib/python3.8/site-packages (1.7.2)

**Design**
https://github.com/BLu85/AES-GCM-128-192-256-bits

**Command**
cd aes-gcm-sx/tb; python3 gcm_testbench.py -m 256 -p 7 -s XS -u 3200000 -j 0 -q 0 -g -c xcelium -b enc -n 1 2>&1 | tee sim.log;

**Makefile fix to get the snapshot to load (remove MAKE_LIB and debugdb)**
ifeq ($(TOPLEVEL_LANG),verilog)
    HDL_SOURCES = $(VERILOG_SOURCES)
    ROOT_LEVEL = $(TOPLEVEL)
    EXTRA_ARGS += -top $(TOPLEVEL)
    ifneq ($(VHDL_SOURCES),)
        HDL_SOURCES += $(VHDL_SOURCES)
        GPI_EXTRA = $(shell cocotb-config --lib-name-path vhpi xcelium):cocotbvhpi_entry_point
    endif
else ifeq ($(TOPLEVEL_LANG),vhdl)
    GPI_EXTRA = $(shell cocotb-config --lib-name-path vhpi xcelium):cocotbvhpi_entry_point
    EXTRA_ARGS += -top $(TOPLEVEL)
    RTL_LIBRARY ?= $(TOPLEVEL)
    HDL_SOURCES = $(VHDL_SOURCES)
    ifneq ($(VERILOG_SOURCES),)
        HDL_SOURCES += $(VERILOG_SOURCES)
    endif
else
    $(error A valid value (verilog or vhdl) was not provided for TOPLEVEL_LANG=$(TOPLEVEL_LANG))
endif

#    MAKE_LIB = -makelib $(RTL_LIBRARY)

# Builds a list of arguments to support VHDL libraries specified in VHDL_SOURCES_*:
LIBS := $(foreach LIB, $(VHDL_LIB_ORDER),-makelib $(LIB) $(VHDL_SOURCES_$(LIB)) -endlib)

$(COCOTB_RESULTS_FILE): $(HDL_SOURCES) $(CUSTOM_COMPILE_DEPS) $(CUSTOM_SIM_DEPS) | $(SIM_BUILD)
	$(RM) $(COCOTB_RESULTS_FILE)

	# Make sure all libs in SOURCES_VHDL_* are mentioned in VHDL_LIB_ORDER and vice versa
	$(foreach LIB, $(VHDL_LIB_ORDER), $(check_vhdl_sources))
	$(foreach SOURCES_VAR, $(filter VHDL_SOURCES_%, $(.VARIABLES)), $(check_lib_order))

	set -o pipefail; \
	MODULE=$(MODULE) TESTCASE=$(TESTCASE) TOPLEVEL=$(TOPLEVEL) GPI_EXTRA=$(GPI_EXTRA) TOPLEVEL_LANG=$(TOPLEVEL_LANG) \
	$(SIM_CMD_PREFIX) $(CMD) -timescale $(COCOTB_HDL_TIMEUNIT)/$(COCOTB_HDL_TIMEPRECISION) \
	$(EXTRA_ARGS) $(GPI_ARGS)  $(INCDIRS) -access +rwc $(LIBS) $(MAKE_LIB) $(PLUSARGS) $(HDL_SOURCES)  2>&1 | tee $(SIM_BUILD)/sim.log

	$(call check_for_results_file)

**Error**
    0.00ns INFO     cocotb.top_aes_gcm                 AES size: XS
     0.00ns ERROR    gpi                                VHPI: Registered callback isn't enabled! Got vhpiStateP=vhpiUndefined(1000)
     0.00ns ERROR    gpi                                Failed to register a value change callback
     1.00ns INFO     cocotb.regression                  test_dut failed
                                                        Traceback (most recent call last):
                                                          File "/mnt/c/aes-gcm/aes-gcm-sx/tb/gcm_test.py", line 96, in test_dut

ktbarrett, 2023-09-22:
Dupe of #1076. This is a known issue. Cadence support isn't great now, but it should be getting better soon now that we have Xcelium running in CI (nearly).

closed 2023-09-22 by ktbarrett

henrikmpedersen00, 2023-10-11:
hi

It doesnt look like its a duplicate of 1076 to me ?

I have pure vhdl sources.

Could you please take a look ?

Thanks

Cheers Henrik

cmarqu, 2023-10-11:
https://github.com/cocotb/cocotb/issues/1076#issuecomment-1351266296 shows that this also happens with pure VHDL designs.