cocotb/cocotb #3424: Cannot override cocotb's logging format

samimia-swks, 2023-09-24:
I am trying to use a consistent logging format for my cocotb and cocotb-test based python testing framework using: 
cocotb 1.7.2
python 3.9.6 (pyenv)
Ubuntu 22.04
I do this by attaching my own handlers and formatters to the root logger and loggers for other modules, including 'cocotb', as the very first thing in my python entry script before even importing cocotb. I also set `proporate=False` for all the child loggers
```
config_logging()  # set up custom handlers/formatters 

import cocotb
# other imports

# rest of code
```

I get the right formatting for my loggers (e.g. 'sim' below) and even the logs produced by cocotb-test using the `cocotb` logger up to point where the sim actually starts: 

```
INFO        sim                   Starting cocotb rtl simulation test_bringup at clk = 3.072MHz ...
INFO        cocotb                Running command: xrun -l ...
INFO        cocotb                TOOL: xrun(64)        22.09-s006: Started on Sep 24, 2023 at 09:20:25 PDT
```

As soon as we get into the @cocotb.test() decorated function however, the logs become a mix of the cocotb formatting and mine, for example: 

```
INFO        cocotb                     0.00ns INFO     cocotb                             Running on xmsim(64) version 22.09-s006
INFO        cocotb                     0.00ns INFO     cocotb                             Running tests with cocotb v1.7.2 from /home/...
INFO        cocotb                     0.00ns INFO     cocotb                             Seeding Python random module with 1695572439
INFO        cocotb                INFO        cocotb.regression     Found test test_bringup.coco_seq
INFO        cocotb                INFO        cocotb.regression     running coco_seq (1/1)
INFO        cocotb                WARNING     pydesign.signals      Assigned value to dout.sdi does not fit in 9 (signed) bits.
```
More concerningly, the WARNING level signal issued by my python module pydesign.signals is 'downgraded' to INFO level...

This is despite the fact that at the very begging of of the @cocotb.test() decorated function, I do everything that is recommended in `log.py` docs and more to try to undo what cocotb did 
```
@cocotb.test()
async def coco_seq(dut):
    # undo the setup cocotb did
    root_logger = logging.getLogger()
    for handler in root_logger.handlers:
        root_logger.removeHandler(handler)
        handler.close()

    logging.basicConfig(level=logging.INFO, force=True)
    logging.setLoggerClass(logging.Logger)
    config_logging() # set up custom handlers/formatters again
```

In https://github.com/cocotb/cocotb/pull/1266 it is suggested that `Another side effect of this change is that logging.basicConfig can now be used to remove the cocotb formatting and handlers.`. Am I doing something wrong here, or has there been a regression?

samimia-swks, 2023-10-12:
Any thoughts here folks?

marlonjames, 2023-10-12:
If you are using `cocotb-test`, then all logs from the simulator subprocess go through the `Simulator._log_pipe()` functions:

https://github.com/themperek/cocotb-test/blob/691ca4917dab79ade86ccedbd8863f337a48c5fe/cocotb_test/simulator.py#L322-L323
Listed here for convenience:
```py3
                asyncio.create_task(self._log_pipe(logging.INFO, p.stdout)),
                asyncio.create_task(self._log_pipe(logging.ERROR, p.stderr)),
```

As you can see, everything that comes over the `stdout` pipe gets logged as `INFO`.

The reason the first few logs coming from cocotb have a different format is because that is the standard log format for cocotb. That is the format that cocotb will use until cocotb's `RegressionManager` loads the Python module containing your test, which then changes the log format.

I think the only way to change the Python logs before your module is loaded would be to provide your own [`PYGPI_ENTRY_POINT`](https://docs.cocotb.org/en/stable/building.html#envvar-PYGPI_ENTRY_POINT) that monkeypatches either `_setup_logging()` or `cocotb.log.default_config()` before calling into cocotb's normal entry point.

Note, however, that even that will not change the `-.--ns INFO     gpi` log format, as those come from the compiled C++ code.

themperek, 2023-10-20:
@marlonjames Is correct. There is no trivial way around this. We have 2 different python instances. 

The way around this (that I was considering) was to forward the log as a machine-readable format to the "runner" and then reinterpret this. One could try to reuse this https://docs.python.org/3/library/logging.handlers.html#datagramhandler
This issue is rather related to cocotb-test than cocotb. We may do it like this in cocotb python runner at some point.

samimia-swks, 2023-10-20:
Thanks for the input. I switched from cocotb-test to the new native test runner in 1.8.1 and things look a lot better. Now the output looks like below. There is still some stuff that is printed with a different format than what I specified, but at least when I call my own loggers inside a coroutine or when I use dut._log(), my formatting is used as expected. 

```log
INFO       11:23:59  sim                    log from my test_*.py before calling cocotb test runner

INFO       11:23:59  cocotb                 INFO: Running command xrun ...
     -.--ns ERROR    gpi                                ..s/cocotb_utils.cpp:68   in utils_dyn_open                  Unable to open lib libpython3.9.so: libpython3.9.so: cannot open shared object file: No such file or directory
     -.--ns INFO     gpi                                ../gpi/GpiCommon.cpp:101  in gpi_print_registered_impl       VPI registered
INFO       11:24:02  cocotb                 INFO: Running command xrun ...
     -.--ns INFO     gpi                                ..mbed/gpi_embed.cpp:105  in set_program_name_in_venv        Using Python virtual environment interpreter at .../.direnv/python-3.9.6/bin/python
     -.--ns INFO     gpi                                ../gpi/GpiCommon.cpp:101  in gpi_print_registered_impl       VPI registered
     -.--ns INFO     gpi                                ../gpi/GpiCommon.cpp:101  in gpi_print_registered_impl       VHPI registered
xcelium> run; exit;
     0.00ns INFO     cocotb                             Running on xmsim(64) version 22.09-s006
     0.00ns INFO     cocotb                             Running tests with cocotb v1.8.1 from .../.direnv/python-3.9.6/lib/python3.9/site-packages/cocotb
     0.00ns INFO     cocotb                             Seeding Python random module with 1697826259
     0.00ns INFO     cocotb.regression                  pytest not found, install it to enable better AssertionError messages

INFO       11:24:20  cocotb.pinname         log from calling dut.pinname._log()

INFO       11:24:21  sim                    log from my test_*.py inside a cocotbcoroutine
```

The new test runner uses print statements instead of calls to logging, which I assume is because it's in experimental status. I got around that in the meanwhile with this nify hack which routes stdout to logging: https://johnpaton.net/posts/redirect-logging/
```
with OutputLogger("cocotb", "INFO"):
        runner = cocotb.runner.get_runner(sim.simulator)
````

This is good enough for me. Should I close this, or leave it open to discuss a more proper solution?

closed 2023-10-20 by themperek