cocotb/cocotb #3426: cocotb / ghdl vpi_get_str: unhandled property for instance name

Naviss, 2023-09-28:
## Description:
When simulation starts, cocotb/ghdl throws a bunch of `vpi_get_str: unhandled property` errors in the log. This seems to happen when cocotb discovers the top entities and passes on top of a module instantiation.

## Environment
- GHDL:
```
GHDL 3.0.0 (3.0.0.r0.g7de967c51) [Dunoon edition]
 Compiled with GNAT Version: 12.3.0
 llvm 15.0.7 code generator
```
- Cocotb : 1.8.0
- OS : Ubuntu 23.04 (Lunar Lobster) 64 bit
- Python : Python 3.11.1 (pipenv / pyenv)

## Log sample :
```
     0.00ns DEBUG    gpi                                End of type=36 iteration
     0.00ns DEBUG    gpi                                No valid type=114 iterator
     0.00ns DEBUG    gpi                                No valid type=48 iterator
     0.00ns DEBUG    gpi                                No valid type=116 iterator
     0.00ns DEBUG    gpi                                No valid type=29 iterator
     0.00ns DEBUG    gpi                                No valid type=25 iterator
     0.00ns DEBUG    gpi                                No valid type=47 iterator
     0.00ns DEBUG    gpi                                No valid type=526 iterator
     0.00ns DEBUG    gpi                                No valid type=618 iterator
     0.00ns DEBUG    gpi                                No valid type=683 iterator
     0.00ns DEBUG    gpi                                No valid type=100 iterator
     0.00ns DEBUG    gpi                                No valid type=34 iterator
     0.00ns DEBUG    gpi                                No valid type=129 iterator
     0.00ns DEBUG    gpi                                No valid type=41 iterator
     0.00ns DEBUG    gpi                                No valid type=103 iterator
     0.00ns DEBUG    gpi                                No valid type=113 iterator
     0.00ns DEBUG    gpi                                No valid type=99 iterator
     0.00ns DEBUG    gpi                                No valid type=31 iterator
     0.00ns DEBUG    gpi                                No valid type=61 iterator
     0.00ns DEBUG    gpi                                No valid type=105 iterator
     0.00ns DEBUG    gpi                                End of type=44 iteration
     0.00ns DEBUG    gpi                                vpi_scan found 'top.sync_generator'
vpi_get_str: unhandled property
vpi_get_str: unhandled property
     0.00ns DEBUG    gpi                                VPI: Created GPI object from type ???(32)
     0.00ns DEBUG    gpi                                Create a native handle
     0.00ns DEBUG    gpi                                Checking top.sync_generator exists
     0.00ns DEBUG    cocotb.sync_generator              Created
     0.00ns DEBUG    gpi                                vpi_scan found 'top.serializerasica'
vpi_get_str: unhandled property
vpi_get_str: unhandled property
     0.00ns DEBUG    gpi                                VPI: Created GPI object from type ???(32)
     0.00ns DEBUG    gpi                                Create a native handle
     0.00ns DEBUG    gpi                                Checking top.serializerasicp exists
     0.00ns DEBUG    cocotb.serializerasicp             Created
...
     0.00ns DEBUG    gpi                                vpi_scan found 'top.serializerasicq'
vpi_get_str: unhandled property
vpi_get_str: unhandled property
     0.00ns DEBUG    gpi                                VPI: Created GPI object from type ???(32)
     0.00ns DEBUG    gpi                                Create a native handle
     0.00ns DEBUG    gpi                                Checking top.serializerasicq exists
     0.00ns DEBUG    cocotb.serializerasicq             Created
     0.00ns DEBUG    gpi                                vpi_scan found 'top.serializerasicr'
vpi_get_str: unhandled property
vpi_get_str: unhandled property
     0.00ns DEBUG    gpi                                VPI: Created GPI object from type ???(32)
     0.00ns DEBUG    gpi                                Create a native handle
     0.00ns DEBUG    gpi                                Checking top.serializerasicr exists
     0.00ns DEBUG    cocotb.serializerasicr             Created
     0.00ns DEBUG    gpi                                vpi_scan found 'top.serializerasics'
```

## VHDL Sample :
```VHDL
     sync_generator : Master_Mode_Pet_Data_Links_Status_Manager port map (
        clk                                                              => clk,

        i_Master_Mode_Events_Stream_Enabled   => '1',
        i_Sync_Counter_Ready                               => enable_sync,

        o_Events_Stream_Enabled_Master_Board  => open,
        o_Sync_Counter_Reset                               => open,
        o_System_Sync_Signal                               => master_sync,
        o_Generate_System_Sync_Signal               => open
    );
...
    serializerasicp : Serializer generic map(
        DATA_WIDTH => ASIC_PACKET_SIZE
        ) port map (
        clock    => clk,
        reset    => rst,
        start    => data_in_p_start,
        data_in  => data_in_p_parallel,
        busy     => data_in_p_busy,
        data_out => i_asic_data_rx(15)
    );
    
    serializerasicq : Serializer generic map(
        DATA_WIDTH => ASIC_PACKET_SIZE
        ) port map (
        clock    => clk,
        reset    => rst,
        start    => data_in_q_start,
        data_in  => data_in_q_parallel,
        busy     => data_in_q_busy,
        data_out => i_asic_data_rx(16)
    );
```

ktbarrett, 2023-09-28:
If I had to bet, it's probably due to GHDL not implementing https://github.com/cocotb/cocotb/blob/1749b85dfb7898db9096d00ee31ec5369c3f5ff6/cocotb/share/lib/vpi/VpiCbHdl.cpp#L214 and https://github.com/cocotb/cocotb/blob/1749b85dfb7898db9096d00ee31ec5369c3f5ff6/cocotb/share/lib/vpi/VpiCbHdl.cpp#L216

The best course of action is to access these properties only when the user requests and then cache them, and you just won't get that info with GHDL. It's not like anyone actually uses them anyways...

Naviss, 2023-09-28:
@ktbarrett Yes, by commenting out the two lines the message disappeared.

ktbarrett, 2023-09-28:
That shouldn't be a terrible difficult change. I'll look into it once the private CI is working again.

PR #3542 (2023-11-30): Don't get definition name or file until necessary
Closes #3426.