cocotb/cocotb #3431: Cocotb driver cannot assign values to the simhandle

aignacio, 2023-10-03:
Hey everyone,
I'm trying to build a simple cocotb bus driver for AHB lite (even knowing that there is one available) and I'm having a strange issue where the values that are assigned to the DUT bus do not get propagated through the simulation. Looks like as if there is nothing driving the DUT, also another coroutine (setup_dut) does not get started by the scheduler if I call it after my AhbLiteMaster object creation.

Here is the unit test I'm trying to run:
```python
import random
import cocotb
import os
import logging
import pytest
import itertools
import sys

from random import randrange
from const import cfg
from cocotb_test.simulator import run
from cocotb.triggers import ClockCycles
from cocotb.regression import TestFactory
from cocotb.clock import Clock
from cocotbext.ahb import AHBLiteMaster

@cocotb.coroutine
async def setup_dut(dut, cycles):
    cocotb.start_soon(Clock(dut.hclk, *cfg.CLK_100MHz).start())
    dut.hresetn.setimmediatevalue(0) 
    await ClockCycles(dut.hclk, cycles) 
    dut.hresetn.setimmediatevalue(1) 

@cocotb.test()
async def run_test(dut):
    ahbMaster = AHBLiteMaster(dut, "slave", dut.hclk)
    await setup_dut(dut, cfg.RST_CYCLES) 
    await ClockCycles(dut.hclk,10)
    await ahbMaster.write(0x123,0xdeadbeef) 
    await ClockCycles(dut.hclk,10)
    await ahbMaster.write(0x456,0xbabebabe) 

def test_ahb_lite():
    """
    Test AHB lite

    Test ID: 1
    """
    module = os.path.splitext(os.path.basename(__file__))[0]
    SIM_BUILD = os.path.join(cfg.TESTS_DIR, f"../run_dir/sim_build_{cfg.SIMULATOR}_{module}")
    extra_args_sim = cfg.EXTRA_ARGS

    run(
        python_search=[cfg.TESTS_DIR],
        verilog_sources=cfg.VERILOG_SOURCES,
        toplevel=cfg.TOPLEVEL,
        module=module,
        sim_build=SIM_BUILD,
        extra_args=extra_args_sim
    )
```

This is the driver class
```python
import cocotb
import enum
import collections.abc

#from cocotb_bus.drivers import BusDriver
# from cocotb_bus.bus import Bus
from cocotb_bus.drivers import BusDriver
from cocotb.handle import SimHandleBase
from cocotb.triggers import RisingEdge
from typing import Any, List, Optional, Sequence, Tuple, Union

class AHBSize(enum.IntEnum):
    BYTE    = 0b000
    HWORD   = 0b001
    WORD    = 0b010
    DWORD   = 0b011
    FWORD   = 0b100
    EWORD   = 0b101

class AHBBurst(enum.IntEnum):
    SINGLE  = 0b000
    INCR    = 0b001
    WRAP4   = 0b010
    INCR4   = 0b011
    WRAP8   = 0b100
    INCR8   = 0b101
    WRAP16  = 0b110
    INCR16  = 0b111

class AHBResp(enum.IntEnum):
    OKAY    = 0b0
    ERROR   = 0b1

class AHBTrans(enum.IntEnum):
    IDLE    = 0b00
    BUSY    = 0b01
    NONSEQ  = 0b10
    SEQ     = 0b11

class AHBWrite(enum.IntEnum):
    WRITE   = 0b0
    READ    = 0b1

class AHBLiteMaster(BusDriver):
    _signals = ["haddr", "hsize", "htrans", "hwdata",
                "hrdata", "hwrite", "hready", "hresp"]

    _optional_signals = ["hburst","hmastlock", "hprot", "hnonsec",
                         "hexcl", "hmaster", "hexokay", "hsel"]

    def __init__(self, entity, name, clock, **kwargs):
        BusDriver.__init__(self, entity, name, clock)
        self.clock = clock
        for signal in self._signals + self._optional_signals:
            if signal not in ["hready", "hresp", "hrdata"]:
                try:
                    default_value = 1
                    getattr(self.bus, signal).setimmediatevalue(default_value)
                except AttributeError:
                    pass
    
    def _convert_size(self, value):
        for hsize in AHBSize:
            if (2**hsize.value) == value:
                return hsize
        raise ValueError(f"No hsize value found for {input_int} number of bytes")

    @staticmethod
    def _check_size(size: int, data_bus_width: int) -> None:
        """Check that the provided transfer size is valid."""
        if size > data_bus_width:
            raise ValueError("Size of the transfer ({} B)"
                             " provided is larger than the bus width "
                             "({} B)".format(size, data_bus_width))
        elif size <= 0 or (size & (size - 1)) != 0:
            raise ValueError("Size must be a positive power of 2")

    @cocotb.coroutine
    async def write(self, address: int, value: Union[int, Sequence[int]], 
                    size: Optional[int] = None) -> None:
        if size is None:
            size = len(self.bus.hwdata) // 8
        else:
            AHBLiteMaster._check_size(size, len(self.bus.hwdata) // 8)
        
        await RisingEdge(self.clock)
        
        # Address phase
        self.bus.haddr.value  = address
        self.bus.htrans.value = AHBTrans(0b10)
        self.bus.hsize.value  = self._convert_size(size) 
        self.bus.hwrite.value = AHBWrite(0b1) 
        await RisingEdge(self.clock)

        # while self.bus.hready.value == 0:
            # await RisingEdge(self.clock)

        # Data phase
        self.bus.hwdata.value = value
        await RisingEdge(self.clock)
```

and here are the waveforms, as you can see clock starts but not the reset
[image]
 if I invert the two lines below, I see the reset getting applied correctly.
```python
async def run_test(dut):
    await setup_dut(dut, cfg.RST_CYCLES) 
    ahbMaster = AHBLiteMaster(dut, "slave", dut.hclk)
```
The DUT design is quite simple as it's just a passthrough from slave -> master IOs. Might be something silly as it is my first time writing a driver, thanks in advance in case someone spot the problem =)

[Info]
**repository** = https://github.com/aignacio/cocotbext-ahb
**cocotb version used** = 1.8.0
**operating system** = Linux 93b8785b4607 5.10.104-linuxkit
**simulator and version** = Verilator 5.012 2023-06-13 rev v5.012-34-g4f13c4d1b
**python version** = 3.10

aignacio, 2023-10-07:
Issue seems to be verilator specific, while working with Icarus it works fine!. Any code that uses cocotb [Bus](https://github.com/cocotb/cocotb-bus/blob/master/src/cocotb_bus/bus.py) class will have similar issue...

Thomasb81, 2023-11-11:
Problem fix in verilator upstream : https://github.com/verilator/verilator/pull/4618.
This issue can be closed.

closed 2023-11-11 by marlonjames