cocotb/cocotb #3432: Incorrect assign from packed to unpacked array

robertszczepanski, 2023-10-04:
When a module consists of packed and unpacked arrays, sometimes only the LSB of the unpacked array is updated in simulations with Verilator.
Assigning values works correctly when:
* source array is packed, sink array is packed,
* source array is unpacked, sink array is packed or unpacked.

Assigning breaks when:
* source array is packed, sink array is unpacked.

This issue has been tested with Icarus and Verilator simulators.
In case of Icarus unpacked arrays are treated as a list of values in Python and these values seem to be correct.
In case of Verilator unpacked arrays are treated as `BinaryValue` which somehow results in **assigning LSB only**.
It has been confirmed that Verilator propagates values correctly internally since waveforms and `$display` statement during runtime contain correct values.

You can see described behaviour in an example below.
RTL design:
```verilog
module test_unpacked_arr #(
    parameter integer WIDTH = 4
) (
    input  logic [WIDTH-1:0] packed_arr_i,
    output logic [WIDTH-1:0] packed_arr_o,
    output logic             unpacked_arr_o[WIDTH]
);

  for (genvar b = 0; b < WIDTH; b++) begin : gen_arr_bits
    assign unpacked_arr_o[b] = packed_arr_i[b];
    assign packed_arr_o[b]   = unpacked_arr_o[b];
  end

endmodule
```
Cocotb script:
```python
import cocotb
from cocotb.triggers import Timer

@cocotb.test()
async def test_unpacked_arr(dut):

    def dump_signals():
        dut._log.info("%s:\t%s", dut.packed_arr_i.name, dut.packed_arr_i.value)
        dut._log.info("%s:\t%s", dut.packed_arr_o.name, dut.packed_arr_o.value)
        dut._log.info("%s:\t%s", dut.unpacked_arr_o.name, dut.unpacked_arr_o.value)
        dut._log.info("")

    for i in range(16):
        dut.packed_arr_i.value = i
        await Timer(1, units="ps")
        dump_signals()
```
Output (Verilator simulator):
```
[...]
0.01ns INFO     cocotb.test_unpacked_arr           packed_arr_i:   1100
0.01ns INFO     cocotb.test_unpacked_arr           packed_arr_o:   1100
0.01ns INFO     cocotb.test_unpacked_arr           unpacked_arr_o: 0000
0.01ns INFO     cocotb.test_unpacked_arr           
0.01ns INFO     cocotb.test_unpacked_arr           packed_arr_i:   1101
0.01ns INFO     cocotb.test_unpacked_arr           packed_arr_o:   1101
0.01ns INFO     cocotb.test_unpacked_arr           unpacked_arr_o: 0001
0.01ns INFO     cocotb.test_unpacked_arr           
0.01ns INFO     cocotb.test_unpacked_arr           packed_arr_i:   1110
0.01ns INFO     cocotb.test_unpacked_arr           packed_arr_o:   1110
0.01ns INFO     cocotb.test_unpacked_arr           unpacked_arr_o: 0000
0.01ns INFO     cocotb.test_unpacked_arr           
0.02ns INFO     cocotb.test_unpacked_arr           packed_arr_i:   1111
0.02ns INFO     cocotb.test_unpacked_arr           packed_arr_o:   1111
0.02ns INFO     cocotb.test_unpacked_arr           unpacked_arr_o: 0001
[...]
```

Additional environment information:
- Cocotb 1.8.0
- Linux Debian 11
- Verilator 5.011 64-bit
- Python 3.10.11

ktbarrett, 2023-10-04:
I want to say this has been reported before, but I wasn't able to find it. Unpacked single dimension arrays are discovered in verilator's VPI as single handles, essentially packed under the hood for performance reasons. Investigation is needed to confirm and see if there is a work-around on our end or if this will need to be pushed to verilator.

ktbarrett, 2023-12-01:
Dupe of #3446.

closed 2023-12-01 by ktbarrett