cocotb/cocotb #3433: Unable to access signals generated in `for` loops

robertszczepanski, 2023-10-04:
Cocotb exposes internal module signals allowing access to them.
However, it is not possible when signals are generated inside for loops.
Cocotb exposes internal module signals allowing access to them, e.g. `logic my_sig` in top level RTL module can be accessed with `cocotb.top.my_sig`.
You can see an example below that there is no `gen_wires` handle in the `dut`.
It has been confirmed that you can access `gen_wires[i].c` in `$display` during Verilator simulation.

RTL design:
```verilog
module test_gen_signals (
    input logic a
);

  wire b;

  assign b = a;

  for (genvar i = 0; i < 5; i++) begin : gen_wires
    wire c;
    assign c = b;
  end

endmodule
```

Cocotb script:
```python
import cocotb

@cocotb.test()
async def test_gen_signals(dut):
    dut.a.value = 1
    dut._log.info("Signal %s: %s", dut.a.name, dut.a.value)
    dut._log.info("Signal %s: %s", dut.b.name, dut.b.value)
    dut._log.info("All: %s", dut.__dict__)
    dut._log.info("Signal %s: %s", dut.gen_wires[0].c.name, dut.gen_wires[0].c.value)
```

Output:
```
0.00ns INFO     cocotb.test_gen_signals            Signal a: 0
0.00ns INFO     cocotb.test_gen_signals            Signal b: 0
0.00ns INFO     cocotb.test_gen_signals            All: {'_handle': <cocotb.simulator.gpi_sim_hdl at 0x5581218757f0>, '_len': None, '_sub_handles': {'a': ModifiableObject(test_gen_signals.a), 'b': ModifiableObject(test_gen_signals.b)}, '_invalid_sub_handles': set(), '_name': 'test_gen_signals', '_type': 'GPI_MODULE', '_fullname': 'test_gen_signals(GPI_MODULE)', '_path': 'test_gen_signals', '_log': <SimBaseLog cocotb.test_gen_signals (INFO)>, '_def_name': '', '_def_file': '', '_discovered': False}
0.00ns INFO     cocotb.regression                  test_gen_signals failed
                                                   Traceback (most recent call last):
                                                     File "test_gen_signals.py", line 10, in test_gen_signals
                                                       dut._log.info("Signal %s: %s", dut.gen_wires[0].c.name, dut.gen_wires[0].c.value)
                                                     File "lib/python3.10/site-packages/cocotb/handle.py", line 370, in __getattr__
                                                       raise AttributeError(f"{self._name} contains no object named {name}")
                                                   AttributeError: test_gen_signals contains no object named gen_wires
```

Additional environment information:
- Cocotb 1.8.0
- Linux Debian 11
- Verilator 5.011 64-bit
- Python 3.10.11

ktbarrett, 2023-10-04:
This works in other simulators. We will need an investigation to figure out what the issue is in verilator. If you can, try another simulator.

themperek, 2023-10-22:
What you get if you do:

You may need to name generate block.
You can check the name by:
```python3
    for thing in Top:
        print(thing)
```

themperek, 2023-10-22:
Can you check what this shows:

```python3
    for thing in dut:
        print(thing._path)
```

marlonjames, 2023-10-23:
This looks like dupe of #1884.

closed 2023-10-30 by ktbarrett