cocotb/cocotb #3434: The testsuite doesn't print the final report

yurivict, 2023-10-10:
It isn't clear in the end if any tests have failed, because there's no final report.

yurivict, 2023-10-10:
I run ```make test``` and the test log ends with:
```
/usr/local/bin/iverilog -o sim_build/sim.vvp -D COCOTB_SIM=1 -s simple_and -f sim_build/cmds.f -g2012  -I./common -I./const_stream  simple_and.sv
rm -f results.xml
MODULE=test_verilog_include_dirs TESTCASE= TOPLEVEL=simple_and TOPLEVEL_LANG=verilog \
         /usr/local/bin/vvp -M /usr/ports/cad/py-cocotb/work-py39/stage/usr/local/lib/python3.9/site-packages/cocotb/libs -m libcocotbvpi_icarus   sim_build/sim.vvp 
     -.--ns INFO     gpi                                ..mbed/gpi_embed.cpp:78   in set_program_name_in_venv        Did not detect Python virtual environment. Using system-wide Python interpreter
     -.--ns INFO     gpi                                ../gpi/GpiCommon.cpp:101  in gpi_print_registered_impl       VPI registered
     0.00ns INFO     cocotb                             Running on Icarus Verilog version 12.0 (stable)
     0.00ns INFO     cocotb                             Running tests with cocotb v1.8.1 from /usr/ports/cad/py-cocotb/work-py39/stage/usr/local/lib/python3.9/site-packages/cocotb
     0.00ns INFO     cocotb                             Seeding Python random module with 1696928206
     0.00ns INFO     cocotb.regression                  Found test test_verilog_include_dirs.test_noop
     0.00ns INFO     cocotb.regression                  running test_noop (1/1)
     0.00ns INFO     cocotb.regression                  test_noop passed
     0.00ns INFO     cocotb.regression                  *********************************************************************************************
                                                        ** TEST                                 STATUS  SIM TIME (ns)  REAL TIME (s)  RATIO (ns/s) **
                                                        *********************************************************************************************
                                                        ** test_verilog_include_dirs.test_noop   PASS           0.00           0.00          4.98  **
                                                        *********************************************************************************************
                                                        ** TESTS=1 PASS=1 FAIL=0 SKIP=0                         0.00           2.36          0.00  **
                                                        *********************************************************************************************
                                                        
gmake[5]: Leaving directory '/usr/ports/cad/py-cocotb/work-py39/cocotb-1.8.1/tests/test_cases/test_verilog_include_dirs'
gmake[4]: Leaving directory '/usr/ports/cad/py-cocotb/work-py39/cocotb-1.8.1/tests/test_cases/test_verilog_include_dirs'
cd test_cases/test_vhdl_access && gmake
gmake[4]: Entering directory '/usr/ports/cad/py-cocotb/work-py39/cocotb-1.8.1/tests/test_cases/test_vhdl_access'
rm -f results.xml
gmake -f Makefile results.xml
gmake[5]: Entering directory '/usr/ports/cad/py-cocotb/work-py39/cocotb-1.8.1/tests/test_cases/test_vhdl_access'
Skipping simulation as only Verilog is supported on simulator=icarus
gmake[5]: Leaving directory '/usr/ports/cad/py-cocotb/work-py39/cocotb-1.8.1/tests/test_cases/test_vhdl_access'
gmake[4]: Leaving directory '/usr/ports/cad/py-cocotb/work-py39/cocotb-1.8.1/tests/test_cases/test_vhdl_access'
cd test_cases/test_vhdl_libraries && gmake
gmake[4]: Entering directory '/usr/ports/cad/py-cocotb/work-py39/cocotb-1.8.1/tests/test_cases/test_vhdl_libraries'
Skipping test since only GHDL, Questa/ModelSim, Xcelium and Incisive are supported
gmake[4]: Leaving directory '/usr/ports/cad/py-cocotb/work-py39/cocotb-1.8.1/tests/test_cases/test_vhdl_libraries'
cd test_cases/test_vhdl_libraries_multiple && gmake
gmake[4]: Entering directory '/usr/ports/cad/py-cocotb/work-py39/cocotb-1.8.1/tests/test_cases/test_vhdl_libraries_multiple'
Skipping test since only GHDL, Questa/ModelSim and Xcelium are supported
gmake[4]: Leaving directory '/usr/ports/cad/py-cocotb/work-py39/cocotb-1.8.1/tests/test_cases/test_vhdl_libraries_multiple'
cd test_cases/test_vhdl_zerovector && gmake
gmake[4]: Entering directory '/usr/ports/cad/py-cocotb/work-py39/cocotb-1.8.1/tests/test_cases/test_vhdl_zerovector'
Skipping simulation as only TOPLEVEL_LANG=vhdl is supported
gmake[4]: 'all' is up to date.
gmake[4]: Leaving directory '/usr/ports/cad/py-cocotb/work-py39/cocotb-1.8.1/tests/test_cases/test_vhdl_zerovector'
gmake[3]: Target 'all' not remade because of errors.
gmake[3]: Leaving directory '/usr/ports/cad/py-cocotb/work-py39/cocotb-1.8.1/tests'
gmake[2]: *** [Makefile:43: do_tests] Error 2
gmake[2]: Leaving directory '/usr/ports/cad/py-cocotb/work-py39/cocotb-1.8.1'
Ran a total of 1 TestSuites and 274 TestCases
gmake[1]: *** [Makefile:56: test] Error 2
gmake[1]: Leaving directory '/usr/ports/cad/py-cocotb/work-py39/cocotb-1.8.1'
*** Error code 2
```

ktbarrett, 2023-10-10:
The testsuite doesn't print when test failures when make errors occur, only when cocotb errors occur. That could be better reported, but honestly the play is to not use make for our regression at some point.

themperek, 2023-10-20:
You may like to use [Python test runner](https://docs.cocotb.org/en/stable/library_reference.html#python-test-runner) will return errors.

closed 2023-11-28 by github-actions[bot]