cocotb/cocotb #3435: The test test_select_testcase_error fails

yurivict, 2023-10-10:
```
gmake[6]: Entering directory '/usr/ports/cad/py-cocotb/work-py39/cocotb-1.8.1/tests/test_cases/test_select_testcase_error'
mkdir -p sim_build
/usr/local/bin/iverilog -o sim_build/sim.vvp -D COCOTB_SIM=1 -s sample_module -f sim_build/cmds.f -g2012   /usr/ports/cad/py-cocotb/work-py39/cocotb-1.8.1/tests/test_cases/test_select_testcase_error/../../../tests/designs/sample_module/sample_module.sv
rm -f results.xml
MODULE=x_tests TESTCASE=y_test TOPLEVEL=sample_module TOPLEVEL_LANG=verilog \
         /usr/local/bin/vvp -M /usr/ports/cad/py-cocotb/work-py39/stage/usr/local/lib/python3.9/site-packages/cocotb/libs -m libcocotbvpi_icarus   sim_build/sim.vvp 
     -.--ns INFO     gpi                                ..mbed/gpi_embed.cpp:78   in set_program_name_in_venv        Did not detect Python virtual environment. Using system-wide Python interpreter
     -.--ns INFO     gpi                                ../gpi/GpiCommon.cpp:101  in gpi_print_registered_impl       VPI registered
     0.00ns INFO     cocotb                             Running on Icarus Verilog version 12.0 (stable)
     0.00ns INFO     cocotb                             Running tests with cocotb v1.8.1 from /usr/ports/cad/py-cocotb/work-py39/stage/usr/local/lib/python3.9/site-packages/cocotb
     0.00ns INFO     cocotb                             Seeding Python random module with 1696908774
     0.00ns ERROR    cocotb.regression                  Requested test(s) ['y_test'] wasn't found in module(s) ['x_tests']
Traceback (most recent call last):
  File "/usr/ports/cad/py-cocotb/work-py39/stage/usr/local/lib/python3.9/site-packages/cocotb/__init__.py", line 233, in _initialise_testbench
    _initialise_testbench_(argv_)
  File "/usr/ports/cad/py-cocotb/work-py39/stage/usr/local/lib/python3.9/site-packages/cocotb/__init__.py", line 316, in _initialise_testbench_
    regression_manager = RegressionManager.from_discovery(top)
  File "/usr/ports/cad/py-cocotb/work-py39/stage/usr/local/lib/python3.9/site-packages/cocotb/regression.py", line 174, in from_discovery
    return cls(dut, tests)
  File "/usr/ports/cad/py-cocotb/work-py39/stage/usr/local/lib/python3.9/site-packages/cocotb/regression.py", line 153, in __init__
    for test in tests:
  File "/usr/ports/cad/py-cocotb/work-py39/stage/usr/local/lib/python3.9/site-packages/cocotb/regression.py", line 250, in _discover_tests
    raise AttributeError("Test(s) %s doesn't exist in %s" % (tests, modules))
AttributeError: Test(s) ['y_test'] doesn't exist in ['x_tests']
Traceback (most recent call last):
  File "/usr/ports/cad/py-cocotb/work-py39/stage/usr/local/lib/python3.9/site-packages/cocotb/__init__.py", line 332, in _sim_event
    scheduler.log.error(msg)
AttributeError: 'NoneType' object has no attribute 'log'
     0.00ns ERROR    gpi                                Passing event to upper layer failed
```

Version: 1.8.1
Python 3.9
FreeBSD 13.2

ktbarrett, 2023-10-10:
Seems to be working as intended as the test is to test the failure to find the test in the module.

closed 2023-10-10 by ktbarrett

yurivict, 2023-10-10:
@ktbarrett 

XFAIL is usually used for expected failures. They shouldn't look like real failures.